pjkundert/cpppo #4: Support for period in tag names

johanfforsberg, 2015-05-04:
For directly accessing tags inside "function blocks" (not sure this is the right terminology) it's necessary to use periods in the tag names, e.g. "ACC_MISC_PLC01_ALMA01.HLimit". Currently this does not seem to work.

pjkundert, 2015-05-04:
I'll look into it today; good catch.

pjkundert, 2015-05-04:
Fixed in branch fix-dotted-tags; try it out, if you want:

    git clone --branch fix-dotted-tags git@github.com:pjkundert/cpppo.git
    cd cpppo
    python setup.py install

I'm doing some work on improving the multi-threaded client API unit tests, so I'll be including these improvements, too, before I push another release -- hopefully by later today.

johanfforsberg, 2015-05-04:
Thanks, that was quick! However, I get this:
```
[johfor@localhost ~]$ python -m cpppo.server.enip.client -a g-b080603-cab03-ctl-plc-01 -vv -p ACC_MISC_PLC01_ALMA01.HLimit
05-04 18:51:49.367        -4883a940 enip.dev DETAIL   __init__                      Logix, Class ID 0x0002, Instance ID   1 created
05-04 18:51:49.367        -4883a940 enip.dev DETAIL   lookup     Class     2/0x0002, Instance   1, Attribute  None ==> None
05-04 18:51:49.367        -4883a940 enip.dev DETAIL   lookup     Class     2/0x0002, Instance   0, Attribute  None ==> None
05-04 18:51:49.367        -4883a940 enip.dev DETAIL   lookup     Class     2/0x0002, Instance   0, Attribute  None ==> None
05-04 18:51:49.367        -4883a940 enip.dev DETAIL   lookup     Class     2/0x0002, Instance   0, Attribute  None ==> meta-Logix
05-04 18:51:49.372        -4883a940 root     DETAIL   __init__   Connect:  Success in   0.005s/  5.000s
05-04 18:51:49.372        -4883a940 enip.cli DETAIL   main       Client Register Rcvd   0.022/  5.000s
05-04 18:51:49.373        -4883a940 enip.cli DETAIL   parse_oper Tag: 'ACC_MISC_PLC01_ALMA01.HLimit' yields Operation: {'path': [{'symbolic': 'ACC_MISC_PLC01_ALMA01.HLimit'}]}
05-04 18:51:49.376        -4883a940 enip.cli DETAIL   unconnecte Client Unconnected Send: {
    "enip.CIP.send_data.CPF.item[0].type_id": 0, 
    "enip.CIP.send_data.CPF.item[1].type_id": 178, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.path.segment[0].class": 6, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.path.segment[1].instance": 1, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.priority": 5, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.path.segment[0].symbolic": "ACC_MISC_PLC01_ALMA01.HLimit", 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.read_tag.elements": 1, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.route_path.segment[0].link": 0, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.route_path.segment[0].port": 1, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.service": 82, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.status": 0, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.timeout_ticks": 157, 
    "enip.CIP.send_data.interface": 0, 
    "enip.CIP.send_data.timeout": 0, 
    "enip.options": 0, 
    "enip.sender_context.input": "bytearray(b'0\\x00\\x00\\x00\\x00\\x00\\x00\\x00')", 
    "enip.session_handle": 5506384, 
    "enip.status": 0
}
05-04 18:51:49.377        -4883a940 root     DETAIL   format_pat Formatted     ACC_MISC_PLC01_ALMA01.HLimit from: [{'symbolic': 'ACC_MISC_PLC01_ALMA01.HLimit'}]
05-04 18:51:49.377        -4883a940 enip.cli DETAIL   issue      Sent   0.004/  5.000s: Single Read  Tag  ACC_MISC_PLC01_ALMA01.HLimit {
    "input": "bytearray(b'L\\x0f\\x91\\x1cACC_MISC_PLC01_ALMA01.HLimit\\x01\\x00')", 
    "path.segment[0].symbolic": "ACC_MISC_PLC01_ALMA01.HLimit", 
    "read_tag.elements": 1, 
    "service": 76
}
05-04 18:51:49.377        -4883a940 enip.cli DETAIL   pipeline   Issuing     0/  1; curr:   0 - last:  -1 ==   1 depth
05-04 18:51:49.395        -4883a940 enip.cli DETAIL   collect    Rcvd   0.016/  5.000s {
    "enip.CIP.send_data.CPF.count": 2, 
    "enip.CIP.send_data.CPF.item[0].length": 0, 
    "enip.CIP.send_data.CPF.item[0].type_id": 0, 
    "enip.CIP.send_data.CPF.item[1].length": 6, 
    "enip.CIP.send_data.CPF.item[1].type_id": 178, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.input": "array('c', '\\xcc\\x00\\x04\\x01\\x00\\x00')", 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.read_tag": true, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.service": 204, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.status": 4, 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.status_ext.data": [
        0
    ], 
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.status_ext.size": 1, 
    "enip.CIP.send_data.interface": 0, 
    "enip.CIP.send_data.timeout": 0, 
    "enip.command": 111, 
    "enip.input": "array('c', '\\x00\\x00\\x00\\x00\\x00\\x00\\x02\\x00\\x00\\x00\\x00\\x00\\xb2\\x00\\x06\\x00\\xcc\\x00\\x04\\x01\\x00\\x00')", 
    "enip.length": 22, 
    "enip.options": 0, 
    "enip.sender_context.input": "array('c', '0\\x00\\x00\\x00\\x00\\x00\\x00\\x00')", 
    "enip.session_handle": 5506384, 
    "enip.status": 0
}
05-04 18:51:49.395        -4883a940 enip.cli DETAIL   pipeline   Completed   1/  1; curr:   0 - last:   0 ==   0 depth
05-04 18:51:49.395        -4883a940 enip.cli DETAIL   validate   Client Single Read  Tag  ACC_MISC_PLC01_ALMA01.HLimit Request: {
    "input": "bytearray(b'L\\x0f\\x91\\x1cACC_MISC_PLC01_ALMA01.HLimit\\x01\\x00')", 
    "path.segment[0].symbolic": "ACC_MISC_PLC01_ALMA01.HLimit", 
    "read_tag.elements": 1, 
    "service": 76
}
05-04 18:51:49.395        -4883a940 enip.cli DETAIL   validate     Yields Reply: {
    "input": "array('c', '\\xcc\\x00\\x04\\x01\\x00\\x00')", 
    "read_tag": true, 
    "service": 204, 
    "status": 4, 
    "status_ext.data": [
        0
    ], 
    "status_ext.size": 1
}
05-04 18:51:49.396        -4883a940 root     DETAIL   format_pat Formatted     ACC_MISC_PLC01_ALMA01.HLimit from: [{'symbolic': 'ACC_MISC_PLC01_ALMA01.HLimit'}]
05-04 18:51:49.396        -4883a940 enip.cli WARNING  validate   Client Single Read  Tag  ACC_MISC_PLC01_ALMA01.HLimit returned non-zero status: Status 4 [0]
05-04 18:51:49.396        -4883a940 enip.cli NORMAL   validate   ACC_MISC_PLC01_ALMA01.HLimit              == None: 'Status 4 [0]'
ACC_MISC_PLC01_ALMA01.HLimit              == None: 'Status 4 [0]'
05-04 18:51:49.396        -4883a940 enip.cli DETAIL   pipeline   Pipelined   1/  1; curr:   0 - last:   0 ==   0 depth
```

pjkundert, 2015-05-04:
OK, the "Controller" (server) side is returning a status code "4"...  Is this a real Rockwell controller, or are you (also) running a simulated Controller (eg. using something like: python -m cpppo.server.enip --print -v ACC_MISC_PLC01_ALMA01.HLimit=INT)?  If so, did you remember to restart the simulated Controller?  The fix was on the "Controller" side of the communications...

johanfforsberg, 2015-05-04:
It's a real Rockwell CompactLogix (I think) PLC.

pjkundert, 2015-05-04:
Ahh.   OK, there must be some other system by which those tags inside function blocks are addressed...   We can support pretty much arbitrarily complex sets of EtherNet/IP CIP "path" elements, so we need to determine which ones to use.  I'll take a look online to see if I can discover what they need to see.

johanfforsberg, 2015-05-05:
Thanks for looking into it. If there's any more information I can supply, just ask!

pjkundert, 2015-05-05:
OK, I've pushed another fix to branch fix-dotted-tags.  It now (correctly) handles generating multiple 'symbolic' path segments, for any Tags containing 1 (or more) '.' symbols.

pjkundert, 2015-05-05:
Found and fixed some multi-threading issues in both Client API and Controller communications simulator.   Aggressive threaded Client tests now pass reliably.

pjkundert, 2015-05-06:
Version 3.6.0 has been pushed, fixing both multi-segment symbolic paths, and some multi-threading issues.

pjkundert, 2015-05-09:
Version 3.6.1 pushed; fixes a problem in the client API's handling of Multiple Service Packets introduced in 3.6.0.

closed 2015-05-09 by pjkundert

johanfforsberg, 2015-05-12:
Amazing, thanks!