pjkundert/cpppo #5: Support for 16-bit instance segment

johanfforsberg, 2015-09-01:
While looking into how Rslinx acquires the list of tags from a PLC I ran across an apparently non-standard CIP Service 0x55 which AFAICT may be used to get the tag names from the PLC. However, for this it uses 16 bit instance segments. It looks like cpppo only supports 8 bit segments, is this something that might change?

The possibility to read the tag names (and particularly the tag addresses) would be highly useful.

pjkundert, 2015-09-01:
Hmm.  It supports 16-bit instance segments in the CIP EPATH.  If you could capture the request/response packets, it would be interesting to see what it looks like.  Unfortunately, the EtherNet/IP CIP documentation leaves much to be desired...

johanfforsberg, 2015-09-01:
Yes, I was probably being too hasty in my conclusions, there's a lot going on there.

I have a wireshark capture of the communication, but it contains a bunch of unrelated traffic so I'll have to clean it up it before sending it to you.

Also found this page that has some very similar capture files: http://condel.cc/~reaper/kool/rakenduslik%20andmeside/labor.html

johanfforsberg, 2015-09-01:
Another find, this .NET library supports listing the tags in the PLC: https://controllogixnet.codeplex.com/

closed 2015-09-29 by pjkundert