pjkundert/cpppo #7: Command Line seems to only send max of

gsingh-cp, 2015-09-18:
Hi

It seems command line usage requires that a number be int16 when being sent.

It is possible to send 32bit aka DINT via command line?

struct.error: short format requires SHRT_MIN <= number <= SHRT_MAX
user$ python -m cpppo.server.enip.client -a 127.0.0.2 MYTAG2=32768

Also, is it possible to have multiple different data types be sent for 1 tag? Furthermore, is there a way to state what type of datatype is being sent?

eg.
python -m cpppo.server.enip.client -a 127.0.0.2 MYTAG2=(int16)32767, (dint)50000

Thanks

pjkundert, 2015-09-18:
Yes, you can specify one of the CIP data types (SINT), (INT), (DINT) or (REAL) -- and now also (BOOL) for the data on the command line.  It applies to all data being written to that tag.  You can write different data types in a separate Tag request, eg:

    python -m cpppo.server.enip.client -a 127.0.0.2 'MYTAG=(INT)32767' 'MYTAG=(DINT)50000' 'MYARRAY[5-6]=(REAL)1.23,4.56'

This is documented in the README.org or README.pdf chapter on the "EtherNet/IP Controller Client".

closed 2015-09-18 by pjkundert

gsingh-cp, 2015-09-18:
Thanks for your reply.

In your above example, should I not have indexes for MYTAG?

python -m cpppo.server.enip.client -a 127.0.0.2 'MYTAG[0]=(INT)32767' 'MYTAG[1]=(DINT)50000' 'MYARRAY[5-6]=(REAL)1.23,4.56'

Also I looked at the documentation for setting for the corresponding server for a message like this, but was unsuccessful.

What would be the correct server syntax for this?

python -m cpppo.server.enip -a 127.0.0.2 --print MYTAG=DINT[1] ...?

Thanks again for your assistance.

pjkundert, 2015-09-18:
Yes, if your MYTAG is an array, then you need to include indices.

If you look here:

https://github.com/pjkundert/cpppo#ethernetip-cip-controller-communications-simulatorclient

there is a description of how to start a simulator with Tags of various data types; you're on the right track.

gsingh-cp, 2015-09-18:
Hi

I am able to start the server to look for different tags with various data types, but not 1 tag with various data types.

Server/Listener (not correct):
python -m cpppo.server.enip -a 127.0.0.2 --print 'MYTAG[0]=INT[1]' 'MYTAG[1]=DINT[1]'

Client:
python -m cpppo.server.enip.client -a 127.0.0.2 'MYTAG[0]=(INT)32767' 'MYTAG[1]=(DINT)500000'

I had a read through the document, seems I am missing something.

Thanks

datasim, 2015-09-18:
No, you must make separate tags; or tag must be if one single type.