pjkundert/cpppo #8: Unable to read the same tag from two different computers

metallo25, 2015-09-30:
Hello,
I am trying to read a tag on a Allen Bradley PLC. I was able to read the tags from Linux PC and there was no issue with that.

An other computer, tried to read the same tag and which made the other computer not being able to read it and vice versa.

This the snippet of the code, I am using on both of these computers.

def read_cip(tag, plc_ip_address):	
	try:	    	    
		with client.connector( host= plc_ip_address) as conn:
			req = conn.read(tag)
			try:
				assert conn.readable( timeout=0.1 ), "Failed to receive reply"
				rpy = next(conn)	
			except AssertionError:	
				print "Need more time!!"	
			return rpy.enip.CIP.send_data.CPF.item[1].unconnected_send.request.read_frag.data[0]

	except socket.error:
	       print "Wrong ip or network error"

please share your thoughts on this.

pjkundert, 2015-09-30:
Interesting.  If two computers try to read, and only one computer at a time can, then perhaps there are limits on the number of "Unconnected" EtherNet/IP CIP connections that the PLC can support?  If it supports only one, then that might explain the issue.

It looks like you are only temporarily setting up the EtherNet/IP CIP connection, performing the read, and then tearing it down.   If the "unconnected" connections are rate-limited by the PLC, then that might explain the issue.

If you could increase the logging level and collect the error response from the failing connection, then we could diagnose exactly the error code being returned by the PLC.

metallo25, 2015-10-14:
Hello Sir,
Your advise helped. I modified to connect only once but read iteratively. I was able to read the same tag from two different computers and no problems in that. Thank you!.

closed 2015-10-14 by metallo25