pjkundert/cpppo #9: Unable to list identity

bafeigum, 2015-10-05:
Using the example EthernetIP-server.py code, I am unable to process the ListIdentity command (0x63). Is this command supported? It should be returning the default Logix identity specified in device.py, but I keep getting an EtherNet/IP error.

Here is the log I am getting.

10-05 22:15:17.659        -49088000 enip.srv NORMAL   main       Delaying all responses by 0.0 seconds
10-05 22:15:17.689        -49088000 enip.srv NORMAL   main       Creating tag: TagName=DINT[1]
10-05 22:15:17.735        -49088000 root     NORMAL   main       EtherNet/IP Simulator: ('', 44818)
10-05 22:15:17.767        -49088000 network  NORMAL   server_mai enip_srv server PID [  981] running on ('', 44818)
10-05 22:15:17.784        -49088000 network  NORMAL   server_mai enip_srv server PID [  981] responding to external done/disable signal
10-05 22:15:20.822        -4cc7aba0 enip.srv NORMAL   enip_srv   EtherNet/IP Server enip_44818 begins serving peer ('192.168.1.99', 44818)
10-05 22:15:20.905        -4cc7aba0 enip.srv DETAIL   enip_srv   Transaction begins
10-05 22:15:20.979        -4cc7aba0 enip.srv DETAIL   enip_srv   Transaction receive after   0.072s (   24 bytes in   0.001/  0.100s)
10-05 22:15:21.019        -4cc7aba0 enip.srv DETAIL   enip_srv                      ( enip_44818.( header.((empty)) recv:    24: 'c\x00\x00\x0...0\x00\x00\x00'
10-05 22:15:21.883        -4cc7aba0 enip.srv DETAIL   enip_srv   Transaction parsed  after   0.978s
10-05 22:15:21.915        -4cc7aba0 enip.dev DETAIL   __init__                   Identity, Class ID 0x0001, Instance ID   1 created
10-05 22:15:21.931        -4cc7aba0 enip.dev DETAIL   lookup     Class     1/0x0001, Instance   1, Attribute  None ==> None
10-05 22:15:21.955        -4cc7aba0 enip.dev DETAIL   lookup     Class     1/0x0001, Instance   0, Attribute  None ==> None
10-05 22:15:21.974        -4cc7aba0 enip.dev DETAIL   lookup     Class     1/0x0001, Instance   0, Attribute  None ==> None
10-05 22:15:22.024        -4cc7aba0 enip.dev DETAIL   lookup     Class     1/0x0001, Instance   0, Attribute  None ==> meta-Identity
10-05 22:15:22.223        -4cc7aba0 enip.dev DETAIL   __init__                      Logix, Class ID 0x0002, Instance ID   1 created
10-05 22:15:22.239        -4cc7aba0 enip.dev DETAIL   lookup     Class     2/0x0002, Instance   1, Attribute  None ==> None
10-05 22:15:22.277        -4cc7aba0 enip.dev DETAIL   lookup     Class     2/0x0002, Instance   0, Attribute  None ==> None
10-05 22:15:22.300        -4cc7aba0 enip.dev DETAIL   lookup     Class     2/0x0002, Instance   0, Attribute  None ==> None
10-05 22:15:22.324        -4cc7aba0 enip.dev DETAIL   lookup     Class     2/0x0002, Instance   0, Attribute  None ==> meta-Logix
10-05 22:15:22.363        -4cc7aba0 enip.dev DETAIL   __init__         Connection_Manager, Class ID 0x0006, Instance ID   1 created
10-05 22:15:22.379        -4cc7aba0 enip.dev DETAIL   lookup     Class     6/0x0006, Instance   1, Attribute  None ==> None
10-05 22:15:22.420        -4cc7aba0 enip.dev DETAIL   lookup     Class     6/0x0006, Instance   0, Attribute  None ==> None
10-05 22:15:22.437        -4cc7aba0 enip.dev DETAIL   lookup     Class     6/0x0006, Instance   0, Attribute  None ==> None
10-05 22:15:22.464        -4cc7aba0 enip.dev DETAIL   lookup     Class     6/0x0006, Instance   0, Attribute  None ==> meta-Connection_Manager
10-05 22:15:22.503        -4cc7aba0 enip.dev DETAIL   __init__             Unknown_Object, Class ID 0x0066, Instance ID   1 created
10-05 22:15:22.530        -4cc7aba0 enip.dev DETAIL   lookup     Class   102/0x0066, Instance   1, Attribute  None ==> None
10-05 22:15:22.564        -4cc7aba0 enip.dev DETAIL   lookup     Class   102/0x0066, Instance   0, Attribute  None ==> None
10-05 22:15:22.583        -4cc7aba0 enip.dev DETAIL   lookup     Class   102/0x0066, Instance   0, Attribute  None ==> None
10-05 22:15:22.608        -4cc7aba0 enip.dev DETAIL   lookup     Class   102/0x0066, Instance   0, Attribute  None ==> meta-Unknown_Object
10-05 22:15:22.733        -4cc7aba0 enip.dev DETAIL   __init__                       UCMM, Class ID 0x9999, Instance ID   1 created
10-05 22:15:22.748        -4cc7aba0 enip.dev DETAIL   lookup     Class 39321/0x9999, Instance   1, Attribute  None ==> None
10-05 22:15:22.774        -4cc7aba0 enip.dev DETAIL   lookup     Class 39321/0x9999, Instance   0, Attribute  None ==> None
10-05 22:15:22.810        -4cc7aba0 enip.dev DETAIL   lookup     Class 39321/0x9999, Instance   0, Attribute  None ==> None
10-05 22:15:22.835        -4cc7aba0 enip.dev DETAIL   lookup     Class 39321/0x9999, Instance   0, Attribute  None ==> meta-UCMM
10-05 22:15:22.878        -4cc7aba0 enip.dev DETAIL   lookup     Class     2/0x0002, Instance   1, Attribute  None ==> Logix
10-05 22:15:22.893        -4cc7aba0 enip.lgx NORMAL   setup                         Logix.TagName       DINT[   1] == 0 Attribute   1 added
10-05 22:15:22.940        -4cc7aba0 enip.dev DETAIL   lookup     Class     2/0x0002, Instance   1, Attribute     1 ==> TagName       DINT[   1] == 0
10-05 22:15:23.078        -4cc7aba0 enip.lgx ERROR    process    EtherNet/IP CIP error at 0 total bytes:
''
-^ (byte 0)

10-05 22:15:23.146        -4cc7aba0 enip.srv ERROR    enip_srv   Failed request: {
    "request.enip.status": 0, 
    "request.enip.sender_context.input": "array('c', '\\x01\\x02\\x03\\x04\\x05\\x06\\x07\\x08')", 
    "request.enip.session_handle": 0, 
    "request.enip.length": 0, 
    "request.enip.command": 99, 
    "request.enip.options": 0, 
    "request.addr": [
        "192.168.1.99", 
        44818
    ]
}
10-05 22:15:23.207        -4cc7aba0 enip.lgx DETAIL   process    EtherNet/IP CIP Request  (Client ('192.168.1.99', 44818)): {
    "addr": [
        "192.168.1.99", 
        44818
    ]
}
10-05 22:15:23.239        -4cc7aba0 enip.dev DETAIL   request    EtherNet/IP (Client ('192.168.1.99', 44818)) Session Terminated: '(Unknown)'
10-05 22:15:23.270        -4cc7aba0 enip.lgx DETAIL   process    EtherNet/IP CIP Response (Client ('192.168.1.99', 44818)): {
    "addr": [
        "192.168.1.99", 
        44818
    ]
}
10-05 22:15:23.331        -4cc7aba0 enip.srv ERROR    enip_srv   EtherNet/IP error at 24 total bytes:
'c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x02\x03\x04\x05\x06\x07\x08\x00\x00\x00\x00'
----------------------------------------------------------------------------------------------^ (byte 24)

Failed with exception:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/cpppo/server/enip/main.py", line 657, in enip_srv
    if enip_process( addr, data=data, **kwds ):
  File "/usr/local/lib/python2.7/dist-packages/cpppo/server/enip/logix.py", line 695, in process
    for i,(m,s) in enumerate( machine.run( path='request.enip', source=source, data=data )):
  File "/usr/local/lib/python2.7/dist-packages/cpppo/automata.py", line 655, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/usr/local/lib/python2.7/dist-packages/cpppo/automata.py", line 1300, in delegate
    raise NonTerminal( "%s sub-machine terminated in a non-terminal state" % ( self ))
NonTerminal: ( CIP.( select.((noop)) ) ) sub-machine terminated in a non-terminal state

10-05 22:15:23.367        -4cc7aba0 enip.srv NORMAL   enip_srv   enip_44818 done; processed   1 request  over    24 bytes/   24 received (0 connections remain)
10-05 22:15:23.398        -4cc7aba0 network  WARNING  run        enip_srv server failure: ( CIP.( select.((noop)) ) ) sub-machine terminated in a non-terminal state
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/cpppo/server/network.py", line 132, in run
    super( server_thread, self ).run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/cpppo/server/enip/main.py", line 657, in enip_srv
    if enip_process( addr, data=data, **kwds ):
  File "/usr/local/lib/python2.7/dist-packages/cpppo/server/enip/logix.py", line 695, in process
    for i,(m,s) in enumerate( machine.run( path='request.enip', source=source, data=data )):
  File "/usr/local/lib/python2.7/dist-packages/cpppo/automata.py", line 655, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/usr/local/lib/python2.7/dist-packages/cpppo/automata.py", line 1300, in delegate
    raise NonTerminal( "%s sub-machine terminated in a non-terminal state" % ( self ))
NonTerminal: ( CIP.( select.((noop)) ) ) sub-machine terminated in a non-terminal state

pjkundert, 2015-10-17:
Unfortunately, cpppo doesn't (yet) support either the List Identity command, or UDP connections.  However, this is something I am working on -- no ETA yet...

pjkundert, 2015-12-11:
The basic List Identity framework is implemented, in branch feature-list-identity.  Comments welcome, as I complete the feature...

pjkundert, 2016-01-30:
Fixed in branch feature-list-identity.  I'll be merging it onto master next week, and pushing Version 3.9.0

closed 2016-01-30 by pjkundert