pjkundert/cpppo #11: Error running the weather.py example

jonwolf-ibisa, 2016-02-20:
Hi, I'm new to **python** and I need to write a program using this **cpppo library** that needs to connect to a server running an Ethernet/IP simulator by Molex in an specific IP address on port 44818.

Using an Ethernet/IP object reader I need to validate that I'm connected and I need to show data from the controller, in this case (RSLinx Server), a first step will be to read object 0x01 that corresponds to the identity. (there's an image attached in this post that shows the Molex Ethernet/IP Simulator)

i wanted to test this weather.py example but I'm getting an error in the line _**from cpppo.server.enip import device, REAL**_ that tells me this:
_**The debugged program raised the exception unhandled ImportError
"No module named 'cpppo'"**_

I've already installed the cpppo library using the terminal command _pip install cpppo_, so, i've got the library installed already.

Could you guys please help with this matter.

Thanks so much in advance.
 

[image]

pjkundert, 2016-02-20:
If you have installed 'cpppo' using 'pip install cpppo', but then your Python cannot find it -- make sure you are using the same Python.  This indicates that there is some issue with your Python installation, not with cpppo.

You might try 'python -m pip install cpppo'.  This will ensure that (at least) the same version of Python is being used to perform the pip install.

If, after a 'pip install whatever', you run python, get a '>>> ' prompt, and try 'import whatever'.  If this doesn't succeed, then the module you just installed didn't get installed correctly.  I'd recommend following some Python getting started documentation to get your feet wet, and work out all of the basic issues with working with Python, and then have another go...

Don't give up.  Working with Python on Windows is especially difficult vs. Mac or Linux.  But, once you've got all the Windows-y issues worked out, it seems to work OK.  Getting the "command line" to work right is especially problematic, but there are work-arounds in the https://github.com/pjkundert/cpppo.git documentation README.org.

Hope that helps.

jonwolf-ibisa, 2016-02-21:
Hi pjkundert, thanks for the quick reply, I've **pip uninstall cpppo** and then **python -m pip install cpppo**  but I'm getting the same error, forgot to mention that I'm working on Virtual Box and Debian 8.3.0. on Windows 7.

[image]

datasim, 2016-02-21:
I would recommend using the command-line.  It's probably an issue with your
IDE, not using the same version of Python as on your Debian installation,
or something.

I'd recommend installing other (simpler) modules, and testing them out,
'til you get it to work.  Then resume working with cpppo.  Follow the
getting started documentation for your IDE, perhaps?

jonwolf-ibisa, 2016-02-21:
you were right datasim, running the program in terminal runs well, the problem is in Eric IDE, i need to configure some sort of path. Thanks

closed 2016-11-30 by pjkundert