pjkundert/cpppo #14: Too many values to unpack

karmicthreat, 2016-03-02:
So I am trying to read a string that is part of an array of strings out of a AB CompactLogix L18ER.  With the structure JugARecipeName STRING[32] and the DATA tag inside is SINT[82]  so 32 strings with a length of 82.  I get a ValueError: too many values to unpack (expected 2) if I try to read one element.  But I can't seem to read JugARepcipeName[1] or JugARecipeName[1].DATA as I just get 'cpppo.automata.NonTerminal: ( data.( select.((noop)) ) ) sub-machine terminated in a non-terminal state' am I doing something wrong to read this? 

Examples below:

python3 -m cpppo.server.enip.client -a 192.168.2.210 -l cpppo.txt --print 'JugARecipeName[1].DATA'
Traceback (most recent call last):
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1609, in <module>
    sys.exit( main() )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1590, in main
    fragment=fragment, printing=printing, timeout=timeout )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1366, in process
    transactions		= list( self.results( operations=operations, **kwds ))
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1358, in results
    for idx,dsc,req,rpy,sts,val in self.operate( operations, **kwds ):
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1353, in operate
    for idx,dsc,req,rpy,sts,val in harvested:
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1243, in validate
    for index,descr,request,reply,status,val in harvested:
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1218, in pipeline
    col		= next( harvester )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1150, in harvest
    issued, self.collect( timeout=timeout )): # must be "lazy" zip!
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1075, in collect
    response,elapsed	= await( self, timeout=timeout )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 832, in await
    for response in cli: # if StopIteration raised immediately, defaults to {} signalling completion
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 509, in __next__
    for mch,sta in engine:
  File "/usr/local/lib/python3.5/site-packages/cpppo/automata.py", line 645, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/usr/local/lib/python3.5/site-packages/cpppo/automata.py", line 1226, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "/usr/local/lib/python3.5/site-packages/cpppo/automata.py", line 645, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/usr/local/lib/python3.5/site-packages/cpppo/automata.py", line 1226, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "/usr/local/lib/python3.5/site-packages/cpppo/automata.py", line 645, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/usr/local/lib/python3.5/site-packages/cpppo/automata.py", line 1291, in delegate
    raise NonTerminal( "%s sub-machine terminated in a non-terminal state" % ( self ))
cpppo.automata.NonTerminal: ( data.( select.((noop)) ) ) sub-machine terminated in a non-terminal state

---------------------------------------------

python3 -m cpppo.server.enip.client -a 192.168.2.210 -l cpppo.txt --print 'JugARecipeName[1].DATA[0]'

Traceback (most recent call last):
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1609, in <module>
    sys.exit( main() )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1590, in main
    fragment=fragment, printing=printing, timeout=timeout )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1366, in process
    transactions		= list( self.results( operations=operations, **kwds ))
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1358, in results
    for idx,dsc,req,rpy,sts,val in self.operate( operations, **kwds ):
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1353, in operate
    for idx,dsc,req,rpy,sts,val in harvested:
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1243, in validate
    for index,descr,request,reply,status,val in harvested:
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1208, in pipeline
    iss		= next( issuer )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 932, in issue
    for op in operations:
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 205, in parse_operations
    seg,elm,cnt		= parse_path_elements( tag )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/device.py", line 314, in parse_path_elements
    elm,_		= elm.split( ']' )
ValueError: too many values to unpack (expected 2)

pjkundert, 2016-03-03:
Richard;

    Thanks for the report.  Yes, the Tag parsing code in Cpppo is presently pretty simplistic.  Asking for elements of arrays of structs is pretty sketchy, at this point.  In the first example, you are successfully asking for the data, but the response parser probably isn't capable of parsing the data response.  In the second example, the simplistic Tag parser is erroneously handling the two indices.

I would recommend trying the first approach, and increasing the verbosity (add -vvv to the CLI command-line).  Then, send me the logging output.  I should be able to diagnose why the response is not being parsed.

I'll work on the Tag parsing code, to properly parse multi-level tags with multiple indices.  In the mean time, you can hand-assemble arbitrarily complex EPATHs (instead of using the Tag parser):

```
python -m cpppo.server.enip.client --print -vvv '@{"symbolic":"JugARecipeName"}/{"element":0}/{"symbolic":"DATA"}'
```

Unfortunately, the element count supplied to the Read Tag [Fragmented] request in these cases is always == 1, so our options are limited to asking for a single element of whatever EPATH you provide...

The EPATH segments that you can ask for (each in a separate JSON dict) are:

```
symbolic
element
class
instance
attribute
port, link
```

karmicthreat, 2016-03-03:
Example below, directly defining the EPATH does work for me:
python3 -m cpppo.server.enip.client -a 192.168.2.210 --print '@{"symbolic":"JugARecipeName"}/{"element":5}/{"symbolic":"DATA"}/{"element":1}'
JugARecipeName.DATA[1][  1-1  ]+  0 == [80]: 'OK'

Verbose example from previous:
python3 -m cpppo.server.enip.client -a 192.168.2.210 --print -vvv 'JugARecipeName[1].DATA[1]'
03-03 14:18:39.361 MainThread enip.cli DETAIL   __init__   Connect:  TCP/IP to ('192.168.2.210', 44818)
03-03 14:18:39.391 MainThread enip.cli DETAIL   cip_send   Client CIP Send: {
    "enip.options": 0,
    "enip.CIP.register.options": 0,
    "enip.CIP.register.protocol_version": 1,
    "enip.sender_context.input": "bytearray(b'\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00')",
    "enip.status": 0,
    "enip.session_handle": 0
}
03-03 14:18:39.391 MainThread enip.cli INFO     cip_send   EtherNet/IP:  24 + CIP:   4 ==  28 bytes total
03-03 14:18:39.391 MainThread enip.cli INFO     send       EtherNet/IP-->   192.168.2.210:44818 send    28: b'e\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00'
03-03 14:18:39.392 MainThread enip.cli INFO     __next__   EtherNet/IP-->   192.168.2.210:44818 rcvd     0: None
03-03 14:18:39.398 MainThread enip.cli INFO     __next__   EtherNet/IP-->   192.168.2.210:44818 rcvd    28: b'e\x00\x04\x00\x01\x00\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00'
03-03 14:18:39.402 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>         enip.command     = 101 (format b'<H' over array('B', [101, 0]))
03-03 14:18:39.403 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>          enip.length     = 4 (format b'<H' over array('B', [4, 0]))
03-03 14:18:39.403 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>  enip.session_handle     = 720897 (format b'<I' over array('B', [1, 0, 11, 0]))
03-03 14:18:39.403 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>          enip.status     = 0 (format b'<I' over array('B', [0, 0, 0, 0]))
03-03 14:18:39.404 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>         enip.options     = 0 (format b'<I' over array('B', [0, 0, 0, 0]))
03-03 14:18:39.404 MainThread enip.cli INFO     __next__   EtherNet/IP      192.168.2.210:44818 done:                         ((enip.((payload.((byte))) -> ((payload.; next byte  28: None      : {'enip': {'options': 0, 'command': 101, 'status': 0, 'session_handle': 720897, 'input': array('B', [1, 0, 0, 0]), 'length': 4, 'sender_context': {'input': array('B', [0, 0, 0, 0, 0, 0, 0, 0])}}, 'peer': ('192.168.2.210', 44818)}
03-03 14:18:39.406 MainThread cpppo    INFO     run                   ( register.( protocol_version.((byte))  -- limit='enip.CIP.register...length' == 4; ending at symbol 4 vs. None
03-03 14:18:39.406 MainThread cpppo    INFO     terminate             ( register.( protocol_version.((byte))  :             => enip.CIP.register.protocol_version     = 1 (format b'<H' over array('B', [1, 0]))
03-03 14:18:39.406 MainThread cpppo    INFO     terminate             ( register.( protocol_version.((byte))  :             => enip.CIP.register.options     = 0 (format b'<H' over array('B', [0, 0]))
03-03 14:18:39.406 MainThread enip.cli INFO     __next__   Returning result: {'enip': {'options': 0, 'command': 101, 'status': 0, 'session_handle': 720897, 'input': array('B', [1, 0, 0, 0]), 'CIP': {'register': {'options': 0, 'protocol_version': 1}}, 'length': 4, 'sender_context': {'input': array('B', [0, 0, 0, 0, 0, 0, 0, 0])}}, 'peer': ('192.168.2.210', 44818)}
03-03 14:18:39.415 MainThread enip.cli NORMAL   __init__   Connect:  Success in   0.054s/  5.000s
03-03 14:18:39.415 MainThread enip.cli DETAIL   main       Client Register Rcvd   0.055/  5.000s
Traceback (most recent call last):
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1609, in <module>
    sys.exit( main() )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1590, in main
    fragment=fragment, printing=printing, timeout=timeout )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1366, in process
    transactions		= list( self.results( operations=operations, **kwds ))
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1358, in results
    for idx,dsc,req,rpy,sts,val in self.operate( operations, **kwds ):
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1353, in operate
    for idx,dsc,req,rpy,sts,val in harvested:
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1243, in validate
    for index,descr,request,reply,status,val in harvested:
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 1208, in pipeline
    iss		= next( issuer )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 932, in issue
    for op in operations:
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/client.py", line 205, in parse_operations
    seg,elm,cnt		= parse_path_elements( tag )
  File "/usr/local/lib/python3.5/site-packages/cpppo/server/enip/device.py", line 314, in parse_path_elements
    elm,_		= elm.split( ']' )
ValueError: too many values to unpack (expected 2)

datasim, 2016-03-07:
> command.((byte)) : => enip.command = 101 (format b' 03-03 14:18:39.403
> enip.length = 4 (format b' 03-03 14:18:39.403 MainThread cpppo INFO
> (format b' 03-03 14:18:39.403 MainThread cpppo INFO terminate ( header.(
> command.((byte)) : => enip.status = 0 (format b' 03-03 14:18:39.404
> enip.options = 0 (format b' 03-03 14:18:39.404 MainThread enip.cli INFO
> (format b' 03-03 14:18:39.406 MainThread cpppo INFO terminate ( register.(

If you clone https://github.com/pjkundert/cpppo.git, and 'git checkout
feature-paths', I've now implemented (hopefully) more complete handling of
more complex paths.   If you 'python setup.py install' it, let me know if
it handles your paths better...

karmicthreat, 2016-03-09:
Seems to work.  I will look closer at it later this week.

datasim, 2016-03-09:
Great; it's a work in progress; I may make changes that don't pass all unit
tests from time to time...

karmicthreat, 2016-03-24:
Had some time.  One other place that I am running into the issue:
python3 -m cpppo.server.enip.client -a 192.168.2.210 -vvv --print 'JugARptStepTime[2][2]'
03-24 14:55:20.278 MainThread enip.cli DETAIL   __init__   Connect:  TCP/IP to ('192.168.2.210', 44818)
03-24 14:55:20.309 MainThread enip.cli DETAIL   cip_send   Client CIP Send: {
    "enip.options": 0,
    "enip.status": 0,
    "enip.sender_context.input": "bytearray(b'\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00')",
    "enip.CIP.register.options": 0,
    "enip.CIP.register.protocol_version": 1,
    "enip.session_handle": 0
}
03-24 14:55:20.309 MainThread enip.cli INFO     cip_send   EtherNet/IP:  24 + CIP:   4 ==  28 bytes total
03-24 14:55:20.309 MainThread enip.cli INFO     send       EtherNet/IP-->   192.168.2.210:44818 send    28: b'e\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00'
03-24 14:55:20.309 MainThread enip.cli INFO     __next__   EtherNet/IP-->   192.168.2.210:44818 rcvd     0: None
03-24 14:55:20.314 MainThread enip.cli INFO     __next__   EtherNet/IP-->   192.168.2.210:44818 rcvd    28: b'e\x00\x04\x00\x01\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00'
03-24 14:55:20.319 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>         enip.command     = 101 (format b'<H' over array('B', [101, 0]))
03-24 14:55:20.319 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>          enip.length     = 4 (format b'<H' over array('B', [4, 0]))
03-24 14:55:20.320 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>  enip.session_handle     = 1376257 (format b'<I' over array('B', [1, 0, 21, 0]))
03-24 14:55:20.320 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>          enip.status     = 0 (format b'<I' over array('B', [0, 0, 0, 0]))
03-24 14:55:20.321 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>         enip.options     = 0 (format b'<I' over array('B', [0, 0, 0, 0]))
03-24 14:55:20.321 MainThread enip.cli INFO     __next__   EtherNet/IP      192.168.2.210:44818 done:                         ((enip.((payload.((byte))) -> ((payload.; next byte  28: None      : {'peer': ('192.168.2.210', 44818), 'enip': {'length': 4, 'input': array('B', [1, 0, 0, 0]), 'command': 101, 'options': 0, 'status': 0, 'session_handle': 1376257, 'sender_context': {'input': array('B', [0, 0, 0, 0, 0, 0, 0, 0])}}}
03-24 14:55:20.323 MainThread cpppo    INFO     run                   ( register.( protocol_version.((byte))  -- limit='enip.CIP.register...length' == 4; ending at symbol 4 vs. None
03-24 14:55:20.323 MainThread cpppo    INFO     terminate             ( register.( protocol_version.((byte))  :             => enip.CIP.register.protocol_version     = 1 (format b'<H' over array('B', [1, 0]))
03-24 14:55:20.323 MainThread cpppo    INFO     terminate             ( register.( protocol_version.((byte))  :             => enip.CIP.register.options     = 0 (format b'<H' over array('B', [0, 0]))
03-24 14:55:20.324 MainThread enip.cli INFO     __next__   Returning result: {'peer': ('192.168.2.210', 44818), 'enip': {'length': 4, 'input': array('B', [1, 0, 0, 0]), 'CIP': {'register': {'protocol_version': 1, 'options': 0}}, 'command': 101, 'options': 0, 'status': 0, 'session_handle': 1376257, 'sender_context': {'input': array('B', [0, 0, 0, 0, 0, 0, 0, 0])}}}
03-24 14:55:20.333 MainThread enip.cli NORMAL   __init__   Connect:  Success in   0.055s/  5.000s
03-24 14:55:20.333 MainThread enip.cli DETAIL   main       Client Register Rcvd   0.055/  5.000s
Traceback (most recent call last):
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.5/site-packages/cpppo-3.9.3-py3.5.egg/cpppo/server/enip/client.py", line 1609, in <module>
    sys.exit( main() )
  File "/usr/local/lib/python3.5/site-packages/cpppo-3.9.3-py3.5.egg/cpppo/server/enip/client.py", line 1590, in main
    fragment=fragment, printing=printing, timeout=timeout )
  File "/usr/local/lib/python3.5/site-packages/cpppo-3.9.3-py3.5.egg/cpppo/server/enip/client.py", line 1366, in process
    transactions		= list( self.results( operations=operations, **kwds ))
  File "/usr/local/lib/python3.5/site-packages/cpppo-3.9.3-py3.5.egg/cpppo/server/enip/client.py", line 1358, in results
    for idx,dsc,req,rpy,sts,val in self.operate( operations, **kwds ):
  File "/usr/local/lib/python3.5/site-packages/cpppo-3.9.3-py3.5.egg/cpppo/server/enip/client.py", line 1353, in operate
    for idx,dsc,req,rpy,sts,val in harvested:
  File "/usr/local/lib/python3.5/site-packages/cpppo-3.9.3-py3.5.egg/cpppo/server/enip/client.py", line 1243, in validate
    for index,descr,request,reply,status,val in harvested:
  File "/usr/local/lib/python3.5/site-packages/cpppo-3.9.3-py3.5.egg/cpppo/server/enip/client.py", line 1208, in pipeline
    iss		= next( issuer )
[... cut ...]
03-24 14:58:12.008 MainThread enip.cli DETAIL   format_pat Formatted                   JugARptStep[2] from: [{u'symbolic': u'JugARptStep'}, {u'element': 2}, {u'element': 2}]
03-24 14:58:12.008 MainThread enip.cli NORMAL   validate         JugARptStep[2][  2-2  ]+  0 == [2]: 'OK'
      JugARptStep[2][  2-2  ]+  0 == [2]: 'OK'
03-24 14:58:12.009 MainThread enip.cli DETAIL   pipeline   Pipelined   1/  1; curr:   0 - last:   0 ==   0 depth
03-24 14:58:12.009 MainThread enip.cli NORMAL   main       Client Tag I/O  Average  51.499 TPS (  0.019s ea).

pjkundert, 2016-11-30:
CIP STRING type is now supported.

closed 2016-11-30 by pjkundert