pjkundert/cpppo #15: can't use 0x4c service to read tag from 1769-L23E-QB1 Ethernet Port

Questio, 2016-04-02:
Hi all, 
     I use cpppio to read a tag named "my_array[0]", and i find the status code replied by my AB PLC  is always "Path Segment Error", the cpppo version i used is v3.9.3, the below is my process:

enip_client -a 128.0.21.107 'my_array[0]'

the request is:
0000   6f 00 2e 00 00 30 02 0b 00 00 00 00 30 00 00 00  o....0......0...
0010   00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00  ................
0020   00 00 00 00 b2 00 1e 00 **52 02 20 06 24 01** 05 9d  ........R. .$...
0030   10 00 4c 06 91 08 6d 79 5f 61 72 72 61 79 28 00  ..L...my_array(.
0040   01 00 01 00 01 00                                ......    

the reply is:
0000   6f 00 16 00 00 30 02 0b 00 00 00 00 30 00 00 00  o....0......0...
0010   00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00  ................
0020   00 00 00 00 b2 00 06 00 cc 00 **04 01** 00 00        ..............
                                     ..
i find the cpppo always use 0x52 service code of CM,  but why not is 0x4c of RM(route manager)?  
can i control only use 0x4c service code of the RM to read/write tags?

pjkundert, 2016-04-02:
The 0x52 request is the "Read Tag" request.  0x4C is the "Read Tag Fragmented" request.  By default, if you don't supply an offset, we'll just use Read Tag.  Evidently, some Rockwell devices don't implement it.

To force "Read Tag Fragmented", use the '--fragment' command-line option, or supply an offset:

    python -m cpppo.server.enip --fragment 'my_array[0]'
    python -m cpppo.server.enip 'my_array[0]+0'

Either way will work.

pjkundert, 2016-04-02:
Sorry, updated my comment.  You can use either:

    enip_client ...
    python -m cpppo.server.enip.client ...

closed 2016-11-30 by pjkundert