pjkundert/cpppo #19: Cannot talk with PLC

zfxw0206, 2016-06-27:
I have a ControlLogix PLC, and want to realize the communication between ROS and this PLC. Previously, I was using slot 0 of PLC, it was fine, but now after changing to slot 4, I could not send the tag to PLC even though I can ping the PLC. Does anybody can help me? Thank you!

pjkundert, 2016-06-27:
You can supply a different route_path when you create the operation.  It
defaults to slot 1.  Change that to port:0, slot:4.  See the documentation
at https://github.com/pjkundert/cpppo

zfxw0206, 2016-06-27:
I found the default route_path in " cpppo/server/enip/main.py" : 
route_path_default = [{'link': 0, 'port': 1}]
Do you mean I should change it to route_path_default = [{'link': 4, 'port': 0}]?

datasim, 2016-06-27:
No, pass link 0, port 1 in the route_path argument to
client.parse_operations.

zfxw0206, 2016-06-27:
I did not see the route_path set in client.parse_operations. Do I need to add it?

datasim, 2016-06-28:
I notice that I haven't updated the documentation!  Sorry about that.  I'll
fix it.

Just pass route_path=... as a keyword parameter to client.parse_operations,
with your custom value.  It will use it (instead of the default) in the
resultant sequence of operations.

zfxw0206, 2016-06-28:
client.parse_operations yield "opr" which is a dictionary. Do you mean add "route_path" in this dictionary? Can you give me more details about where to add "route_path" and how this parameter looks link? Is it "opr['route_path'] =   [{'link': 4, 'port': 0}] (I am using slot 4)"? Thank you!

datasim, 2016-06-28:
Just pass another keyword parameter when you call parse_operations( ...,
route_path={'link': 4, 'port': 0} <http://i%20am%20using%20slot%204/> )

It takes all remaining unrecognized keywords and adds then to each
operation it yields.

zfxw0206, 2016-06-28:
I have a code named "write_to_plc", in this code, I didn't call parse_operations, but called "client.main". previously, my code is like this:
**result = client.main(["-a", plc_ip_address, combinedTag])**

Now, I want to add "--route-path" like: 
**result = client.main(["-a", plc_ip_address, combinedTag]["--route-path", [{'link': 4, 'port': 1}]])**

Then, I got the following error:
**[ERROR] [WallTime: 1467144731.410220] bad callback: <bound method CallbackListener.ros_heartbeat_callback of <__main__.CallbackListener instance at 0x7faf9584f710>>
Traceback (most recent call last):
    result = client.main(["-a", plc_ip_address, combinedTag]["--route-path", [{'link': 4, 'port': 1}]])
TypeError: list indices must be integers, not tuple**

Can I set the route path in "client.main"? If I can, how to fix this error?
Thank you!

datasim, 2016-06-28:
> *[ERROR] [WallTime: 1467144731.410220] bad callback: > Traceback (most

Ah, if you are using cpppo.server.enip client.main, then you need to supply
the --route-path as valid JSON.  Only double-quoted dictionary keys are
allowed.  So, change your code to ... *[ "--route-path", '[{"link": 4,
"port": 1}]' ]*

I assumed you were using the client API directly, and therefore calling
client.parse_operations directly in your code.

zfxw0206, 2016-06-28:
I changed the route path to '[{"link": 4, "port": 1}]', but it still doesn't work. 
Except for setting route path, is there other option to write to plc when the plc uses slot 4?

datasim, 2016-06-28:
I'm almost certain that it should be link:0, port:4.

zfxw0206, 2016-06-29:
Because I am calling client.main and client.main calls parse_operations, I want to print route path in client.main to check if I set it successfully. I add the print after **line 1518 of client.py** , but it print nothing.  Can you tell me a way to display the route path, then I can check it. Thank you!

datasim, 2016-06-29:
I would recommend adding some -v options to increase verbosity; the logging
should display the route_path used.

zfxw0206, 2016-06-29:
I did something wrong yesterday. Now I am using **'[{"link": 4, "port": 1}]'**, it is working. Thank you very much!

closed 2016-06-29 by zfxw0206