pjkundert/cpppo #23: proxy.write not implemented

patrickjmcd, 2016-09-29:
I'm attempting to read/write from a Rockwell E300, but I'm unable to get the write half working.

```
python3 -m cpppo.server.enip.get_attribute -S -a 10.20.4.9 "@0x09/1/3" "@0x09/1/3=(BOOL)1" -v
09-29 09:47:07.815 MainThread enip.cli NORMAL   __init__   Connect:  Success in   0.031s/  5.000s
Thu Sep 29 09:47:07 2016:   0: Single G_A_S      @0x0009/1/3 == [0]
Thu Sep 29 09:47:07 2016:   1: Single S_A_S      @0x0009/1/3 == None
09-29 09:47:07.834 MainThread enip.get NORMAL   main         2 requests in   0.015s at pipeline depth  0; 129.873 TPS
```

Does anyone have experience with this?

datasim, 2016-09-29:
Ya, I've been working on generalizing the cpppo methods to more cleanly
handle writes. It works fine (you can try it out using
cpppo.server.enip.client via the Command-Line Interface (see docs).
Remember to specify the data type as DINT, eg ...
'some_tag[0-3]=(DINT)0,1,2,3'

I have an experimental branch called 'feature-proxy-write' that is a
work-in-progress, to get the cpppo.server.enip.get_attribute 'proxy'
interface to cleanly handle both reading and writing. You could take a look
at that.

I have been much too busy to work on this; some Cpppo "Support" contracts
could help that... ;)

pjkundert, 2016-09-30:
Also; when using the Get/Set Attribute Single EtherNet/IP CIP request -- you must use raw SINT data of the size of the target attribute.  No CIP data type can be used; they are not supported by Get/Set Attribute Single/All.

So, I would recommend trying the above call with "@0x09/1/3=(SINT)1" instead, and see what happens.  If you add a  `-vvv` option, we can see what the CIP error code is.

patrickjmcd, 2016-09-30:
I'll give that a shot tomorrow and report back. Thanks for the quick response.

pjkundert, 2016-11-30:
Writing data via the cpppo.server.enip.get_attribute proxy API should now work.

closed 2016-11-30 by pjkundert