pjkundert/cpppo #28: Connection to Control Logix via 1756-EN2TR

jpautomation, 2017-01-11:
This looks like a great bit of code.

Can I just confirm that it should be possible I to route through an 1756-EN2TR Ethernet module to the Control Logix PLC eg 5572S? I've run the enip.client to list connections and I can only see the Ethernet module not the CPU.

Thanks for your help

jpautomation, 2017-01-12:
Answering my own question..

Yes you can.

What threw me was that a product na,e of "1756-EN2TR" is reported from the Identity query not the CPU part number but reading a tag from the ethernet module IP address still works

jpautomation, 2017-01-12:
Ok wonder if someone can help me here. ]

I can read tags fine with the client.connector.parse_operations with:

host = "192.168.1.5"     # Controller IP address
tags = [ "TestTag_DINT" ,"dummy", "TestString2[1-10]" ]
with client.connector( host=host ) as conn:
    for index,descr,op,reply,status,value in conn.pipeline(
            operations=client.parse_operations( tags ), depth=2 ):
        print( "%s: %20s: %s" % ( time.ctime(), descr, value ))

but when I use connector,read:
value = conn.read ("TestTag_DINT")

I just get the attributes of the tag returned to value. 

And when I try to write it doesn't change the tag value:
req1 = conn.write( "TestTag_DINT", data=1234 )

What am I doing wrong. Tags are read/write and can be changed from Studio 5000.

Any help appreciated. Thanks in advance

pjkundert, 2017-01-27:
I would recommend using the cpppo.server.enip.get_attribute "proxy" API.  The cpppo.server.enip.client "connector" API is very low-level; it is designed for issuing and harvesting a stream of requests to/from a device, in a pipelined fashion.  Usually, this is a bit lower level than you'd want to be operating...

See server/enip/poll_example_many_with_write.py for an example of how to repetitively poll values from a device, and occasionally write new values without interrupting the polling.  Of course, you can just perform the writes without the polling...

Hope that helps!

EavenSheets, 2017-04-20:
@pjkundert Do you have any examples of just writing to a device? I can read from a device quite easily using a symbolic tag, but I am having trouble trying to write to a CompactLogix PLC using get_attribute proxy.

pjkundert, 2017-04-21:
Take a look at the documentation at https://github.com/pjkundert/cpppo, and search for "proxy.write".  You can use the proxy API to write, by supplying a "Tag = <value>", instead of just "Tag".

closed 2017-04-21 by pjkundert