pjkundert/cpppo #29: read() from client.connector is no longer working

metallo25, 2017-02-28:
Hello SIr,
I am using the latest cpppo release and read() from client.py is no longer functional. I checked the same program working in 2.6.1.

Here is a code snippet of that:

```
while True:
		try:
			with client.connector( host= plc_ip_address) as conn:
				while True:
					time_prev=time.time()
					req = conn.read("tag_1")
					try:
						assert conn.readable( timeout=2.0 ), "Failed to receive reply"
						rpy = next(conn)
						d.data.append(rpy.enip.CIP.send_data.CPF.item[1].unconnected_send.request.read_frag.data[0])#0
					except AssertionError:	
						print "Response timed out!! Tearing Connection and Reconnecting!!!!!"
						break
					except AttributeError:
						print "Tag  not written:::Will try again::"
						break
```

I am getting the message as an attribute error. But the same program gets the tag on computer having the old cpppo 2.6.1 I guess. Any thoughts?.

pjkundert, 2017-03-13:
I just tried reproducing this, using cpppo version 3.9.5 on a Mac:
```
import time
from repr import repr

import cpppo
from cpppo.server.enip import client

plc_ip_address          ='127.0.0.1'
d                       = cpppo.dotdict( { 'data': [] } )

with client.connector( host= plc_ip_address) as conn:
    while True:
        time_prev=time.time()
        req = conn.read("tag_1")
        try:
            assert conn.readable( timeout=2.0 ), "Failed to receive reply"
            rpy = next(conn)
            d.data.append(rpy.enip.CIP.send_data.CPF.item[1].unconnected_send.request.read_frag.data[0])#0
        except AssertionError:
            print "Response timed out!! Tearing Connection and Reconnecting!!!!!"
            break
        except AttributeError:
            print "Tag  not written:::Will try again::"
            break
        print "%5d: %s" % ( len( d.data ), d.data )
        time.sleep( 1 )
```

I first ran a simulator:
```
$ python -m cpppo.server.enip -v tag_1=DINT[100]
```

Then, I ran the above program: 

```
$ python src/cpppo_connector_read.py
    1: [0]
    2: [0, 0]
    3: [0, 0, 0]
    4: [0, 0, 0, 0]
    5: [0, 0, 0, 0, 0]
...
```

Is this still happening for you?

metallo25, 2017-03-13:
Hello Sir,
The same program works on simulated plc controller. But does not work when given the IP address of an actual PLC. But the programs function well on 3.9.4 on communicating with actual PLC

metallo25, 2017-03-13:
Is there any significant updates which I would miss out on 3.9.5?. I tested the programs on 3.9.4 and it works great!!. So please advise. Thank you!!

pjkundert, 2017-03-15:
Support for the CIP STRING type, and cpppo.server.enip.get_attribute proxy.write support (the ability to write via the proxy API).  And other minor fixes.

closed 2017-03-15 by pjkundert