pjkundert/cpppo #30: Browsing Tags

karmicthreat, 2017-04-21:
Not really an issue, so much as a question.  Is there any facility to browse tags and get a list of all tags on the PLC?

pjkundert, 2017-04-21:
Unfortunately, no publicly known method to do that.  This appears to be a Rockwell-private functionality (since, clearly, RSLogix can do it!).  But, not documented that I've been able to find.

closed 2017-04-21 by pjkundert

derp7331, 2020-03-19:
This library seems to be able to do this; https://github.com/dmroeder/pylogix

https://github.com/dmroeder/pylogix/blob/251f12a681ebafc38fa1cdb0a3ae1c916f0047a4/examples/23_get_all_tags.py

Since I don't have the PLC's in questions, I'm not able to test it. But thought it would be good to share it anyways.

karmicthreat, 2020-03-25:
The pertinent code in question that builds the tag list request https://github.com/dmroeder/pylogix/blob/0c8a3de5e3b32afe6297bc5dd5d87a16975ca12e/pylogix/eip.py#L1411