pjkundert/cpppo #31: Unable to get proxy.write to write to CompactLogix L32E

EavenSheets, 2017-04-21:
I've been working on getting this library to properly write to my PLC for the past couple of days. Now this is not a device I myself programmed, but something I am trying to control with a python script. I have connected directly to it and the images below are from RSLinx Data Monitoring on the program to grab the tag names. I can read information from the device with the proxy.read interface. For example below:

```python
with via:
    params = via.parameter_substitution('Local:1:I.Data')
    value, = via.read(params)

# with input on device is switched to off
> value
[0]

# with input on device is switched to on
> value
[1]
```

```python
with via:
    params = via.parameter_substitution('Local:1:I.Data = (SINT)1')
    value, = via.write(params, checking=True)

---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-115-5ddb72bb3208> in <module>()
      1 with via:
      2     params = via.parameter_substitution('Local:1:I.Data = (SINT)1')
----> 3     value, = via.write(params, checking=True)
      4

/Users/esheets/.virtualenvs/c3po/lib/python3.6/site-packages/cpppo/server/enip/get_attribute.py in read(self, attributes, printing, checking)
    438         if checking:
    439             assert len( bad ) == 0, \
--> 440                 "read failed to access %d attributes: %s" % ( len( bad ), ', '.join( bad ))
    441
    442     def read_details( self, attributes ):

AssertionError: read failed to access 1 attributes: Local:1:I.Data = (SINT)1: status 5
```

Is there something I am missing? Based on these images that tag is classified as Read/Write so I should be able to change this value from 0 -> 1, and thus turning on the PLC actions.

[image]
[image]

EavenSheets, 2017-04-21:
The only thing I can think is that I am either not understanding the data types properly or there is some setting on the PLC restricting me from actually writing to the device although it says Read/Write.

pjkundert, 2017-04-25:
I would recommend trying to write to a Tag specified in a more "global" namespace of the PLC, first, to validate that everything is working.

The name 'Local:1:I.Data' will be parsed into two "symbolic" names: 'Local:1:I' and 'Data' by Cpppo; I don't know if that is correct; perhaps these type of Tags require the rest of the names to be also broken into separate "symbolic" address segments.  Perhaps try specfiying 'Local.1.I.Data', to force Cpppo to parse each name component?  Just a guess.

Better yet, capture a wireshark trace of some other EtherNet/IP CIP application reading/writing that attribute, and look at what kind of addressing it is using to accomplish the I/O.  This will tell us how to formulate the request.

Unfortunately, the documentation for EtherNet/IP CIP is so bad, that this is how I've had to develop pretty much the entire Cpppo project...