pjkundert/cpppo #32: Request context mismatched error (Poller)

simone201, 2017-05-02:
I am using the Poller class to read periodically a few tags from an AB Micro820 PLC, so running CPPPO in simple mode without send_path and route_path as false.
Usually it works just fine, reading some default tags into the PLC one at a time (as this plc doesn't support multi request), but sometimes happens this kind of error that i cannot debug at all or figure out why is it happening:

https://pastebin.com/kakCBdJW

In this time the library doesn't read correctly the tags and throws an error. This kind of problem happens randomly and more frequently if i reduce the timing of the poller down to half a second (usually working at 1s reading).

Any idea what's happening there?
Thanks in advance!

pjkundert, 2017-05-03:
At a brief glance, it appears the the Ethernet/IP CIP connection stream is getting a mis-matched response to an issued request; almost as if it issues a request, times out on a response, and then issues another request and receives an non-matching response.

I would need to see the entire Python application to ensure that you are (properly) discarding the proxy session on all errors (ie. never catch exceptions and ignore them within the lifespan of the proxy.  If a request fails (ie. times out) -- you *must* discard the session and establish a new one!

So, check that, and let me look at the full code for how you handle the proxy...

simone201, 2017-05-04:
Here's che code snippet that configures and runs the poller object, on a Micro820 PLC in Run mode over Ethernet cabled connection (even directly to my pc, same thing happens)
https://pastebin.com/CvjG4SDi

It reads out 12 tags which are the digital inputs, and nothing more...