pjkundert/cpppo #34: Performance Optimization

nisarg09, 2017-06-05:
Hi, I'm trying to read 100 "TEXT" tag using pipeline on one of the embedded device, I'm getting performance of 8 to 11 sec by varying depth and multiple values. Can you please help with the proper combination of depth and multiple to get good performance. Also I want to do the same for reading 1000 SCADA tags.

pjkundert, 2017-06-05:
Just enough depth to keep the device saturated with requests.  Mostly, depth is required due to slow (high latency) communications channels.

The multiple flag allows Cpppo to aggregate multiple requests into a single transmitted CIP request.  If the target device can handle it, it may increase the total number of I/O operations that can be handled, per second, by reducing the over head of decoding multiple separate requests.

So, simply by experimenting you should be able to find the optimal settings for your device.

Finally -- Cpppo is 100% pure Python.  Even the low-level parser is Python.  So, I have seen some noticable speed-ups by using PyPy, especially if your program runs for long periods of time (ie. performs some requests, waits then performs the operation again (without shutting down and restarting))

closed 2017-06-05 by pjkundert

JesseJ12345, 2020-06-12:
@pjkundert , you said **The multiple flag allows Cpppo to aggregate multiple requests into a single transmitted CIP request** , but how?  Could you show a demo?

JesseJ12345, 2020-06-12:
We found the communications is fast, but between the last receive and next send(read multiple tags ), it used **2.596616 - 2.443908 =‬ 0.152708 S**   , as the pictures shows below:
[image]

and we also check the source code and test it, we found it is to slow in following function, could you help us?
[image]