pjkundert/cpppo #36: "await" will become reserved word

chrisgilmerproj, 2017-06-28:
I'm using `cpppo==3.9.7` with python 3.6.1's asyncio and just got this warning when running my program with `python -Wdefault myscript.py`:

```
[...]/.venv/lib/python3.6/site-packages/cpppo/server/enip/client.py:853: DeprecationWarning: 'async' and 'await' will become reserved keywords in Python 3.7
  def await( cli, timeout=None ):
```

It comes from this line:

https://github.com/pjkundert/cpppo/blob/master/server/enip/client.py#L853

Renaming this would help future proof the repo.

pjkundert, 2017-11-24:
Checkout the next release, available via pip/pip2/pip3:

     pip install --upgrade git+https://github.com/pjkundert/cpppo.git@feature-multipath

It allows specifying single or multi-hop route_path in the form <port>/<link> or JSON list: '["<port>/<link>", ...]' (eg. route_path=1/0, or route_path='["1/3","2/10.0.1.2"]')

chrisgilmerproj, 2017-11-27:
That looks great, love the deprecation warning.  I'm excited for the new release.  I appreciate you working on this.

mjbrisebois, 2019-07-07:
This is now fixed in version 4.0.3

closed 2019-07-07 by mjbrisebois