pjkundert/cpppo #37: Error"sub-machine terminated in a non-terminal state"

nisargpandyadattus, 2017-07-25:
I'm using CPPPO for reading string data from the PLC. Here is the Error I'm getting while trying to read string tag.
**raise NonTerminal( "%s sub-machine terminated in a non-terminal state" % ( self ))
cpppo.automata.NonTerminal: ( data.( select.((noop)) ) ) sub-machine terminated in a non-terminal state**
Here is the detailed log of the request:
debian@DattusBBB84BC:~/code_repository/branches/EthernetIP/python-codes/source$ python3 -m cpppo.server.enip.client --print -vvv 'CIP1_Data_Log_Text[0]' -a 10.10.135.120
07-25 09:33:28.784 MainThread enip.cli DETAIL   __init__   Connect:  TCP/IP to ('10.10.135.120', 44818)
07-25 09:33:29.171 MainThread enip.cli DETAIL   cip_send   Client CIP Send: {
    "enip.status": 0,
    "enip.sender_context.input": "bytearray(b'\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00')",
    "enip.session_handle": 0,
    "enip.CIP.register.protocol_version": 1,
    "enip.CIP.register.options": 0,
    "enip.options": 0
}
07-25 09:33:29.178 MainThread enip.cli INFO     cip_send   EtherNet/IP:  24 + CIP:   4 ==  28 bytes total
07-25 09:33:29.180 MainThread enip.cli INFO     send       EtherNet/IP-->   10.10.135.120:44818 send    28: b'e\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00'
07-25 09:33:29.183 MainThread enip.cli INFO     __next__   EtherNet/IP-->   10.10.135.120:44818 rcvd    28: b'e\x00\x04\x00\x01\x00:\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00'
07-25 09:33:29.188 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>         enip.command     = 101 (format b'<H' over array('B', [101, 0]))
07-25 09:33:29.192 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>          enip.length     = 4 (format b'<H' over array('B', [4, 0]))
07-25 09:33:29.197 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>  enip.session_handle     = 3801089 (format b'<I' over array('B', [1, 0, 58, 0]))
07-25 09:33:29.202 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>          enip.status     = 0 (format b'<I' over array('B', [0, 0, 0, 0]))
07-25 09:33:29.210 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>         enip.options     = 0 (format b'<I' over array('B', [0, 0, 0, 0]))
07-25 09:33:29.216 MainThread enip.cli INFO     __next__   EtherNet/IP      10.10.135.120:44818 done:                         ((enip.((payload.((byte))) -> ((payload.; next byte  28: None      : {'peer': ('10.10.135.120', 44818), 'enip': {'status': 0, 'sender_context': {'input': array('B', [0, 0, 0, 0, 0, 0, 0, 0])}, 'session_handle': 3801089, 'length': 4, 'command': 101, 'input': array('B', [1, 0, 0, 0]), 'options': 0}}
07-25 09:33:29.220 MainThread cpppo    INFO     run                   ( register.( protocol_version.((byte))  -- limit='enip.CIP.register...length' == 4; ending at symbol 4 vs. None
07-25 09:33:29.225 MainThread cpppo    INFO     terminate             ( register.( protocol_version.((byte))  :             => enip.CIP.register.protocol_version     = 1 (format b'<H' over array('B', [1, 0]))
07-25 09:33:29.229 MainThread cpppo    INFO     terminate             ( register.( protocol_version.((byte))  :             => enip.CIP.register.options     = 0 (format b'<H' over array('B', [0, 0]))
07-25 09:33:29.232 MainThread enip.cli INFO     __next__   Returning result: {'peer': ('10.10.135.120', 44818), 'enip': {'status': 0, 'sender_context': {'input': array('B', [0, 0, 0, 0, 0, 0, 0, 0])}, 'session_handle': 3801089, 'length': 4, 'CIP': {'register': {'protocol_version': 1, 'options': 0}}, 'command': 101, 'input': array('B', [1, 0, 0, 0]), 'options': 0}}
07-25 09:33:29.235 MainThread enip.cli NORMAL   __init__   Connect:  Success in   0.451s/  5.000s
07-25 09:33:29.237 MainThread enip.cli DETAIL   main       Client Register Rcvd   0.453/  5.000s
07-25 09:33:29.240 MainThread enip.cli DETAIL   parse_oper Tag: 'CIP1_Data_Log_Text[0]' yields Operation: {'path': [{'symbolic': 'CIP1_Data_Log_Text'}, {'element': 0}]}.update({'route_path': None, 'send_path': None})
07-25 09:33:29.247 MainThread enip.cli DETAIL   cip_send   Client CIP Send: {
    "enip.status": 0,
    "enip.sender_context.input": "bytearray(b'0\\x00\\x00\\x00\\x00\\x00\\x00\\x00')",
    "enip.session_handle": 3801089,
    "enip.CIP.send_data.interface": 0,
    "enip.CIP.send_data.CPF.item[0].type_id": 0,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.status": 0,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.route_path.segment[0].link": 0,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.route_path.segment[0].port": 1,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.service": 82,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.timeout_ticks": 157,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.path.segment[0].symbolic": "CIP1_Data_Log_Text",
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.path.segment[1].element": 0,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.read_tag.elements": 1,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.service": 76,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.request.input": "bytearray(b'L\\x0b\\x91\\x12CIP1_Data_Log_Text(\\x00\\x01\\x00')",
    "enip.CIP.send_data.CPF.item[1].unconnected_send.priority": 5,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.path.segment[0].class": 6,
    "enip.CIP.send_data.CPF.item[1].unconnected_send.path.segment[1].instance": 1,
    "enip.CIP.send_data.CPF.item[1].type_id": 178,
    "enip.CIP.send_data.timeout": 0,
    "enip.options": 0
}
07-25 09:33:29.257 MainThread enip.cli INFO     cip_send   EtherNet/IP:  24 + CIP:  56 ==  80 bytes total
07-25 09:33:29.259 MainThread enip.cli INFO     send       EtherNet/IP-->   10.10.135.120:44818 send    80: b'o\x008\x00\x01\x00:\x00\x00\x00\x00\x000\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\xb2\x00(\x00R\x02 \x06$\x01\x05\x9d\x1a\x00L\x0b\x91\x12CIP1_Data_Log_Text(\x00\x01\x00\x01\x00\x01\x00'
07-25 09:33:29.262 MainThread enip.cli DETAIL   format_pat Formatted            CIP1_Data_Log_Text[0] from: [{'symbolic': 'CIP1_Data_Log_Text'}, {'element': 0}]
07-25 09:33:29.264 MainThread enip.cli DETAIL   issue      Sent   0.019/  5.000s: Single Read  Tag  CIP1_Data_Log_Text[0] {
    "path.segment[0].symbolic": "CIP1_Data_Log_Text",
    "path.segment[1].element": 0,
    "read_tag.elements": 1,
    "service": 76,
    "input": "bytearray(b'L\\x0b\\x91\\x12CIP1_Data_Log_Text(\\x00\\x01\\x00')"
}
07-25 09:33:29.269 MainThread enip.cli DETAIL   issue      Sending  1 (Context       b'0')
07-25 09:33:29.271 MainThread enip.cli DETAIL   pipeline   Issuing     0/  1; curr:   0 - last:  -1 ==   1 depth
07-25 09:33:29.274 MainThread enip.cli INFO     __next__   EtherNet/IP-->   10.10.135.120:44818 rcvd   136: b'o\x00p\x00\x01\x00:\x00\x00\x00\x00\x000\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\xb2\x00`\x00\xcc\x00\x00\x00\xa0\x02\xce\x0f\x1f\x00\x00\x00CIP PATH #2 WASH W/HOT SANITIZE\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
07-25 09:33:29.278 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>         enip.command     = 111 (format b'<H' over array('B', [111, 0]))
07-25 09:33:29.282 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>          enip.length     = 112 (format b'<H' over array('B', [112, 0]))
07-25 09:33:29.287 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>  enip.session_handle     = 3801089 (format b'<I' over array('B', [1, 0, 58, 0]))
07-25 09:33:29.291 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>          enip.status     = 0 (format b'<I' over array('B', [0, 0, 0, 0]))
[... 168 lines cut ...]
07-25 09:46:53.754 MainThread enip.cli DETAIL   pipeline   Completed   1/  1; curr:   0 - last:   0 ==   0 depth
[... cut ...]
**07-25 09:46:53.772 MainThread enip.cli NORMAL   validate   CIP1_Data_Log_Text.DATA[82][ 82-82 ]+  0 == None: 'Status 5 '**
CIP1_Data_Log_Text.DATA[82][ 82-82 ]+  0 == None: 'Status 5 '
07-25 09:46:53.775 MainThread enip.cli DETAIL   pipeline   Pipelined   1/  1; curr:   0 - last:   0 ==   0 depth
07-25 09:46:53.777 MainThread enip.cli NORMAL   main       Client Tag I/O  Average   4.705 TPS (  0.213s ea).
Here as you can see, the output I'm getting is None and Status is 5.
Can you please help me solving this issue?

iodizedfate, 2017-09-18:
I was having the same issue as you and found this work around.

Instead of ``` CIP1_Data_Log_Text[0].DATA/82 ``` try and use ``` CIP1_Data_Log_Text[0].DATA[0]*82 ``` where instead of 82 use the number that is presented when you call ``` CIP1_Data_Log_Text[0].LEN ```. This should capture what you're looking for in an array of ASCII values of your string. I don't know how this will help in the command line portion of this but if you are using a script, then all you have to do is assign it to a variable and then use 

```
desired_string = ''.join(chr(i) for i in value)
```
where value is the variable holding the ASCII array and desired_string is the output you're looking for.

nisargpandyadattus, 2017-10-17:
@iodizedfate Thanks..I tried this and it seems working..

closed 2017-10-17 by nisargpandyadattus