pjkundert/cpppo #38: Server is not using specified "ip_address" in cpppo.cfg

mjbrisebois, 2017-09-08:
I'm using `configparser` to generate this config file.  All the Identity attributes get loaded and work but **"ip_address"** is still using the default `127.0.0.1`.

```
[Identity]
vendor number = 1
device type = 19
product code number = 45
product revision = 711
status word = 0
serial number = 497368838
product name = 1756-IB32/B DCIN
state = 0
configuration consistency value = 0
heartbeat interval = 0

[TCPIP]
interface configuration = {'ip_address': '172.17.0.2', 'network_mask': '255.255.0.0', 'dns_primary': '8.8.8.8', 'dns_secondary': '8.8.4.4', 'domain_name': 'example.com'}
host name = controller
```

pjkundert, 2017-09-08:
I think all of those other values are the "defaults", except the one you have for "ip_address".  

I wonder if *none* of those values are being loaded; the supplied string is not valid JSON (must use double-quotes, not single quotes).  Try changing other values and see if they actually work.  Try changing the quote character...

pjkundert, 2017-09-08:
Ya, I think that's it.  And, if so, we need to investigate why proper logging of that JSON failure is not happening.