pjkundert/cpppo #39: Problem addressing an assembly object

beville, 2017-09-09:
Hey there,

I'm new to Ethernet/IP and CIP, and recently discovered this project.  Fantastic stuff.  Just playing with your CLI apps has taught me a lot.  My plan is to use your python libraries as test harness for a C/C++ app that will use another open-source project, OpENer (https://github.com/EIPStackGroup/OpENer), which I'm sure you're aware of.

Right now, I'm in the stage of making sure all the parts talk to each other.  The OpENer library has a test app that presents a handful of assembly objects, with some instance IDs like 0x64, 0x96, and 0x97.  I'm trying a command like the following
`python -m cpppo.server.enip.get_attribute -a 192.168.3.13  "@4/100/3" --send-path=""`

Unfortunately, I always get back this:
`Fri Sep  8 17:32:08 2017:   0: Single G_A_S      @0x0004/100/3 == None`

I'm also using the Molex windows tool to send "Get Attribute Single" and "Set Attribute Single" commands to 4/100/3, and that *does* succeed, so I know that the OpENer app can be responsive.  Also, as a sanity check, using the following identity query works as well:
`python -m cpppo.server.enip.client -a 192.168.3.13 -i`

Do I need to something special to talk to an assembly object?

pjkundert, 2017-09-09:
Try adding the -S option, and get rid if the --route-path, so it only uses
"simple" (non Routing) CIP requests.  Like for a MicroLogix, vs. for a
C*Logix controller.

beville, 2017-09-09:
That did it! I thought I had tried that '-S' option, but maybe I had another broken option or config at the time.

Thanks!

closed 2017-09-09 by pjkundert

LucaFR1, 2022-08-03:
Hi, sorry this is propably not the right place to ask, but since you (beville) did exactly what i want to, you could maybe give me a few hints regarding going from python to C++.
I did some simple tests with this VERY nice python project to make sure i can caommunicate with my linear stage via Ethernet/IP. But now I have to move to a C++ implementation. Would you recomment using OpENer? Is it as simple as this python project, where i just have to state the IP as well as the objet, isntance and attribute identifiers?

befuddled-dude, 2022-08-03:
@LucaFR1 Oh, wow, 5 years later!  I can't share much about OpENer, unfortunately.  That project/contract never got past the research phase, so I never did any real development beyond a basic POC app, and I don't have any of that code or notes anymore.  Best of luck.

beville, 2022-08-03:
Oh wow, five years later!  Sorry, that project/contract never went anywhere, and I don't have any notes or code left over, and don't really remember much about it.  Best of luck!