pjkundert/cpppo #43: Probing for Tags

nisargpandyadattus, 2017-10-27:
Hi, 
Is there any facility available like probing in which we can query for tags from the PLC directly to get to know which are the tags configured?

pjkundert, 2017-11-13:
Nope.  Like every other part of the protocol, we'd need to sniff the activity using Wireshark, and then reverse-engineer the protocol elements used.  Not documented, in any of the thousands of pages of documentation that I've read.

kyle-github, 2017-11-27:
It turns out that AB **does** document this.  In their own "special" way.  I got this from a user of my library and I thought I would pass along the joy for the season:

[Logix5000 Data Access Programming Manual](http://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf)

Have fun.  I love that the tag instance numbers come back in 32-bit quantities, but you can only use 16-bit instance IDs when requesting the next batch.

kyle-github, 2017-11-29:
Attribute 7 gives the size of one element and attribute 8 gives the array dimensions (all 3 at once each as a u32).

pjkundert, 2017-11-29:
Thanks for that!  I'll have a look.  Pull requests accepted!!