pjkundert/cpppo #44: Problem reading/writing STRING data types from 1769-L30ER/A Logix5330ER PLC

davkuzmin, 2017-11-06:
I'm using the client to read/write DINT tags like so:
`python -m cpppo.server.enip.client -a myaddress "myDINTtag"`
`python -m cpppo.server.enip.client -a myaddress "myDINTtag=(DINT)1234"`

However, when I attempt to read a string like so:
`python -m cpppo.server.enip.client -a myaddress "mySTRINGtag"`
I get [this error](https://pastebin.com/0xrMTASm) output.
When I try to write a string value like so:
`python -m cpppo.server.enip.client -a myaddress "mySTRINGtag=(STRING)asdf"`
I get the following result:
`mySTRINGtag   <= ['asdf']: 'Status 255 [8445]'`

Am I doing anything obviously wrong? How can proceed debugging this issue?

pjkundert, 2017-11-24:
It looks like the STRING data type is not being parsed properly on the first example:

    python -m cpppo.server.enip.client -a myaddress "mySTRINGtag"

If you could add -vvv and make the request, and give me the logging output, it would tell me where the parser is failing.

Likewise, the device is rejecting an attempt to write a CIP "STRING" to the specified Tag.  I am unsure what the extended status code 8445 means; as with almost everything EtherNet/IP CIP, there is little to no documentation available (correction: there is vast quantities of documentation available from the ODVA: but it is almost entirely useless).

smotero92, 2019-02-13:
I'm finding a similar issue when trying to read a STRING type tag. I use 
```
python -m cpppo.server.enip.client -a "192.168.1.101" "PVString[0]
```
and get the output
```
Traceback (most recent call last):
  File "C:\PYTHON36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\PYTHON36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 1649, in <module>
    sys.exit( main() )
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 1630, in main
    fragment=fragment, printing=printing, timeout=timeout )
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 1406, in process
    transactions                = list( self.results( operations=operations, **kwds ))
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 1398, in results
    for idx,dsc,req,rpy,sts,val in self.operate( operations, **kwds ):
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 1393, in operate
    for idx,dsc,req,rpy,sts,val in harvested:
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 1283, in validate
    for index,descr,request,reply,status,val in harvested:
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 1258, in pipeline
    col         = next( harvester )
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 1190, in harvest
    issued, self.collect( timeout=timeout )): # must be "lazy" zip!
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 1115, in collect
    response,elapsed    = await( self, timeout=timeout )
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 872, in await
    for response in cli: # if StopIteration raised immediately, defaults to {} signalling completion
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\server\enip\client.py", line 546, in __next__
    for mch,sta in engine:
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\automata.py", line 645, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\automata.py", line 1226, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\automata.py", line 645, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\automata.py", line 1226, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\automata.py", line 645, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "C:\Users\sotero\python\TPM_CMMS\venv\lib\site-packages\cpppo\automata.py", line 1291, in delegate
    raise NonTerminal( "%s sub-machine terminated in a non-terminal state" % ( self ))
cpppo.automata.NonTerminal: ( data.( select.((noop)) ) ) sub-machine terminated in a non-terminal state
```

then i use 
```
python -m cpppo.server.enip.client -vvv "192.168.1.101" "PVString[0]"
```
as you request from OP and get 
```
02-13 11:35:33.151 MainThread enip.cli DETAIL   __init__   Connect:  TCP/IP to ('localhost', 44818)
02-13 11:35:34.197 MainThread enip.cli DETAIL   cip_send   Client CIP Send: {
    "enip.session_handle": 0,
    "enip.options": 0,
    "enip.status": 0,
    "enip.sender_context.input": "bytearray(b'\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00')",
    "enip.CIP.register.options": 0,
    "enip.CIP.register.protocol_version": 1
}
02-13 11:35:34.198 MainThread enip.cli INFO     cip_send   EtherNet/IP:  24 + CIP:   4 ==  28 bytes total
02-13 11:35:34.199 MainThread enip.cli INFO     send       EtherNet/IP-->       localhost:44818 send    28: b'e\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\
x00\x01\x00\x00\x00'
02-13 11:35:34.200 MainThread enip.cli INFO     __next__   EtherNet/IP-->       localhost:44818 rcvd    28: b'e\x00\x04\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\
x00\x01\x00\x00\x00'
02-13 11:35:34.200 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>         enip.command     = 101 (format b'<H' over array('B', [101, 0]))
02-13 11:35:34.201 MainThread cpppo    INFO     terminate                        ( header.( command.((byte))  :             =>          enip.length     = 4 (format b'<H' over array('B', [4, 0]))
[... 250 lines cut ...]
02-13 11:35:34.256 MainThread enip.cli DETAIL   validate     Yields Reply: {
    "input": "array('B', [204, 0, 8, 0])",
    "service": 204,
    "status": 8,
    "status_ext.size": 0,
    "read_tag": true
}
02-13 11:35:34.256 MainThread enip.cli DETAIL   format_pat Formatted                      PVString[0] from: [{'symbolic': 'PVString'}, {'element': 0}]
02-13 11:35:34.257 MainThread enip.cli WARNING  validate   Client Single Read  Tag  PVString[0] returned non-zero status: Status 8
[... cut ...]
02-13 11:35:34.259 MainThread enip.cli DETAIL   pipeline   Pipelined   2/  2; curr:   1 - last:   1 ==   0 depth
02-13 11:35:34.259 MainThread enip.cli NORMAL   main       Client Tag I/O  Average  37.762 TPS (  0.026s ea).
```

Any Ideas?
I've gotten the same results trying this
```
host = "192.168.1.101"
tags = ["PVString[0]"]
        with client.connector(host=host) as conn:
            for index, descr, op, reply, status, value in conn.pipeline(
                    operations=client.parse_operations(tags), depth=3):
                print("%20s: %s" % (descr, value))
```