pjkundert/cpppo #46: unreadable tags

andrew199999, 2017-11-23:
Hi
i am approaching this library to have an idea on how to read tags,
In my case i have a L73 processor (clx) in a rack and i would like to read a tag from it. 

The plc module is in a rack with other modules in it, (some i/o cards, a EN2T module). The ip address i can reach is the one of the ethernet module.
My quesitons are: 

1)how i connect directly to the CLX i need? can't i specify rack/port of the backplane?

2) (if 1 goes well)if i have a tag XYZ, can i just put "XYZ" as tagname or i need to specify "ProjectName.TagName" when asking for it?

thanks

pjkundert, 2017-11-24:
Yes, you can indeed specify the backplane slot to perform I/O operations on.

The default 'route_path' is [{"port": 1, "link": 0}].  This indicates the Backplane ("port": 1), slot 0 ("link": 0).  You can specify any other slot X ("link": X), or even multi-hop 'route_path', by putting multiple entries in the route_path list.

I am working on a fairly major update to Cpppo, to provide 3.7 compatiblity (ie. due to the new 'await' keyword), and to provide better support for arbitrary route_path in requests.  You can access this new version (presently 4.0.1) using pip/pip2/pip3:

     pip install --upgrade git+https://github.com/pjkundert/cpppo.git@feature-multipath

andrew199999, 2017-11-27:
Thanks for this, the path selection works fine!

pjkundert, 2017-11-28:
Take a look at: https://github.com/pjkundert/cpppo/blob/master/server/enip/poll_example_many.py

andrew199999, 2017-11-30:
Great!
Next question that comes naturally is: is there any way to get all the tags at once from a PLC?