pjkundert/cpppo #48: Weather.py example local test

checor, 2018-01-17:
Hello,

I want to be able to test the example weather.py from the same machine.

First, I opened the source and put my own API key. Then I run the example using `python weather.py`, and then, in another terminal, tried to get the temperature of a city using tags, with something like  `python -m cpppo.server.enip.get_attribute "New York"`, but it seems that `__getitem__` is never called.

Should I use a specific CIP Class/Instance/Attribute to obtain weather of a city? Or another tool?

Many thanks in advance, your library has helped me to introduce myself to this protocol.

shangarnithi, 2019-02-15:
hi Checor,

I also want to try the same in my linux machine , but couldn't succeeded . 
Please let me know did you find any clue ? and how to test it in two terminals ?

Thanks in advance :)

checor, 2019-02-15:
hello shangarnithi,

As far I can recall, I couldn't make it work locally, but I could try another time.

shangarnithi, 2019-02-20:
@checor Thanks for the reply. 
If it is between two different system in same network also fine for me ...

Thanks