pjkundert/cpppo #55: Incompatibility with Python 3.7

scontini76, 2018-07-13:
citing https://docs.python.org/3/whatsnew/3.7.html

> [async](https://docs.python.org/3/reference/compound_stmts.html#async) and [await](https://docs.python.org/3/reference/expressions.html#await) are now reserved keywords.

that brings many issues, for example:

```
>>> import cpppo.server.enip.get_attribute
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/site-packages/cpppo/server/enip/get_attribute.py", line 376
    rsp,ela		= client.await( connection, timeout=self.timeout )
                          ^
SyntaxError: invalid syntax
```

NOTE: if you have suggestion on how change that function name and you are agree to introduce an API change to accomodate Python 3.7 I can take care of that change.

MPW-waters, 2019-07-02:
Is this fix (#57) planned to be incorporated in order to support 3.7?

mjbrisebois, 2019-07-07:
This is now fixed in version 4.0.4

closed 2019-07-07 by mjbrisebois