pjkundert/cpppo #58: Question: Can this library be used to get/set attributes for any CIP required objects?

rajatkmehta, 2018-08-28:
If I want to get any attribute id for any specif objects shown below, can i do that using this library?

0x01 | Identity Objects | CIP Required Object
0x04 | Assembly Object | CIP object for Drive Device
0x06 | Connection Manager Object | Communication Object 
0XF4 | Port object | Communication Object
0xA0 | Vendor Parameters Object | Vendor Specific
0xF5 | TCP/IP Interface Object | CIP Required Object
0x02 | Message Router Object | Communication Object
0xF4 | Port Object | Communication Object
0xF6 | Ethernet Link Object | CIP Required Object

If yes, please give a code snippet using your library for the same.

je13fr, 2018-08-30:
Hello,
I would say yes as all of this are explicit EIP communication
What you mention if I m not wrong are application layer to finally access EIP objects/attributes/instances
Jerome

rajatkmehta, 2018-09-03:
@je13fr yes correct, i want one example of code to see how any explicit communication can be done for any type of object.

pjkundert, 2018-09-03:
Explicit connections are a subset of EtherNet/IP CIP communications that are officialy supported by Cpppo.  Implicit connections (using Forward Open), usually used for real-time data connections between PLCs on the plant floor, are *not* yet officially supported (although, they are available in a branch of Cpppo, if you want to try them out).

The simplest method for gaining access to a "routing" EtherNet/IP CIP device (such as a C*Logix PLC) is to use the Cpppo cpppo.server.enip.get_attribute 'proxy' object.  Use the 'proxy_simple' to access a non-routing EtherNet/IP CIP device, such as an A-B Powerflex (or pretty much any other device with only one EtherNet/IP CIP port and no backplane).

For example, if the device is a C*Logix (routing) PLC and contains an 4-SINT (4-byte) assembly (class 0x04) object at instance 1, attribute 100 (of course, replace the '127.0.0.1' with the IP address of your PLC):
```
>>> from cpppo.server.enip.get_attribute import proxy
value, = proxy( '127.0.0.1' ).read( '@0x04/1/100=(SINT)1,2,3,4' )
>>> value
True
>>> value, = proxy( '127.0.0.1' ).read( '@0x04/1/100[0-3]' )
>>> value
[1, 2, 3, 4]
>>>
```

je13fr, 2018-09-04:
Hello,
I m very interested on implicit exchange, how to get access of un-official branch ?
Thanks
Best regards
Jerome

pjkundert, 2018-09-04:
The branch is feature-cpppo; look into cpppo/server/enip/client.py for the 'implicit' class.

rajatkmehta, 2018-09-05:
@je13fr please send the path of to access that branch feature-cpppo

je13fr, 2018-09-05:
Hello,
Well, my understanding from  pjkundert is that it's already in this cpppo release but not supported.
I went through cpppo/server/enip/client.py but it's unclear where the implicit class is located, I m still looking at.
@pjkundert : could you point out class names ?
Thanks by advance
Jerome

rajatkmehta, 2018-09-28:
@je13fr did you get for implicit?
What about Explicit, suppose if we need to get attribute for IdentityObject then?

rajatkmehta, 2018-12-20:
@je13fr Did you find out something workable for implicit messaging?
Does this library support Explicit Connected messaging as well?
Request you to please respond!

rajatkmehta, 2018-12-20:
@pjkundert Does this library support Explicit Connected messaging as well?
And not able to find out the class for implicit messaging in path you have mentioned cpppo/server/enip/client.py
Request you to please respond!

rajatkmehta, 2019-01-03:
@pjkundert I have tried below example and getting response from my device about List of Identity
Now my question is, if i change class name as 'f5' for TCP/IP object then still showing response of List of Identity only. How to get response for other objects?? Please suggest.

from cpppo.server.enip.get_attribute import *

cn = proxy_simple('<my ip address of the device>')

attributes = [("@1/1", 'SSTRING')] # CIP Device Name
try:
    value = cn.read(attributes=attributes)
    print(value)
except Exception as exc:
    cn.close_gateway(exc)