pjkundert/cpppo #59: How to manage ENIP server with "connected" getAttributesAll()

je13fr, 2018-08-30:
Hello,

I m starting cpppo server node as follow, with this cppo.cfg:
---------------------------------------------------------------------------------------
[Identity]
Vendor Number			= 243
Device Type			= 14
Product Code Number		= 54
Product Revision		= 2836
Status Word			= 12640
Serial Number			= PMELNX0101
Product Name			= raspiPartner
State				= 255
Configuration Consistency Value	= 0
Heartbeat Interval		= 0

[TCPIP]
Interface Status		= 2  # (Hardware configured)
Configuration Capability	= 48 # (Config Settable, Hardware Settable)
Configuration Control		= 0  # Statically assigned IP
Host Name			= PMELNX1010-RaspI

Path to Physical Link		= [
        { "class":     100 },
        { "instance":    1 }
]
---------------------------- EndFile--------------------------------

My command line to start server is : 
python -m cpppo.server.enip --verbose

When I select "connected mode I got this error with a tool and with my client:

[image]

How to start server to handle this or how can I overcome this issue ?

I attach wireshark trace for further details
The trace contains:
1/ unconnected getAttributeAll() with Hilsher tools
2/ then connected error with Hilsher tools (screenshot) 
and finally the 
3/ getAttributesAll from my client (DTM client).

Thanks by advance for your support :)

Best regards
Je13fr

[ENIP_ConnectedGetAttributesAll.zip](https://github.com/pjkundert/cpppo/files/2336167/ENIP_ConnectedGetAttributesAll.zip)

rajatkmehta, 2018-12-25:
@je13fr Can you upload your code as well for both Unconnected and Connected messaging.?

je13fr, 2018-12-25:
Hello Rajatmehta,
As descrbed on my post, I start server from command line and this one is written in my post as well as config.
I do not chnage anything on server mode and I don't know where to change "connection attributes from connected or unconnected.
My client is a DTM that I don't manage the source as I access it through an SDK but interfaces abstract a lot EIP protocol unfortunatelly, so I have to deal with lot's of retro engineering to understand how to handle client request from server side.
And in server there is quite a lot to understand t know where to get this (or I m not good enough which is probably the point)
Jerome