pjkundert/cpppo #61: Question: Does the library support PCCC over CIP messages?

bmhh, 2018-09-11:
Hi,

I am currently evaluating a set of libraries and have not found a way to read "legacy"-tags like 'N7:0' from the PLC.

Am I overlooking something?

Regards

KaloyanR, 2019-02-07:
Using an Device EDS file to find the Class, Instance and Attribute (CIA) values for a parameter.

EDS files are simple text files used by network configuration tools to help you identify products and easily commission them on a network*.

KaloyanR, 2019-02-07:
https://rockwellautomation.custhelp.com/app/answers/detail/a_id/47397

pjkundert, 2019-02-07:
You must indeed tunnel PCCC over EtherNet/IP CIP.  I have an experimental branch that I was working on getting this implemented, but I ran out of time to complete it; few seemed to take an interest (financially) in obtaining this functionality, so I had to prioritize other things...