pjkundert/cpppo #67: NameError: name 'item' is not defined

alifuk, 2019-02-06:
In one terminal I run command:
 `enip_server SCADA=INT[1000]`

From another terminal I run command:
` enip_client -i`
works fine, but command 
`enip_client SCADA`
fails.

Server output:
`02-06 13:38:11.535   enip_srv network  WARNING  run        enip_srv server failure: name 'item' is not defined
Traceback (most recent call last):
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\server\network.py", line 151, in run
    super( server_thread, self ).run()
  File "c:\users\ays\appdata\local\programs\python\python36\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\server\enip\main.py", line 631, in enip_srv
    enip_srv_tcp( conn, addr, name=name, enip_process=enip_process, delay=delay, **kwds )
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\server\enip\main.py", line 817, in enip_srv_tcp
    if enip_process( addr, data=data, **kwds ):
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\server\enip\logix.py", line 777, in process
    data.response               = dotdict( data.request )
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\dotdict.py", line 79, in __init__
    self.update( *args, **kwds )
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\dotdict.py", line 85, in update
    self.__setitem__( key, val )
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\dotdict.py", line 160, in __setitem__
    target[rest]        = value
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\dotdict.py", line 160, in __setitem__
    target[rest]        = value
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\dotdict.py", line 160, in __setitem__
    target[rest]        = value
  [Previous line repeated 1 more time]
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\dotdict.py", line 155, in __setitem__
    target          = eval( mine, {'__builtins__':{}}, self )
  File "<string>", line 1, in <module>
NameError: name 'item' is not defined`

Client output:
`Traceback (most recent call last):
  File "c:\users\ays\appdata\local\programs\python\python36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\ays\appdata\local\programs\python\python36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\AYS\AppData\Local\Programs\Python\Python36\Scripts\enip_client.exe\__main__.py", line 9, in <module>
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\server\enip\client.py", line 1630, in main
    fragment=fragment, printing=printing, timeout=timeout )
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\server\enip\client.py", line 1406, in process
    transactions                = list( self.results( operations=operations, **kwds ))
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\server\enip\client.py", line 1398, in results
    for idx,dsc,req,rpy,sts,val in self.operate( operations, **kwds ):
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\server\enip\client.py", line 1393, in operate
    for idx,dsc,req,rpy,sts,val in harvested:
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\server\enip\client.py", line 1283, in validate
    for index,descr,request,reply,status,val in harvested:
  File "c:\users\ays\appdata\local\programs\python\python36\lib\site-packages\cpppo\server\enip\client.py", line 1270, in pipeline
    complete, requests )
AssertionError: Communication ceased before harvesting all pipeline responses:   0/  1`

How can I obtain value of Tag 'SCADA'?

jhbrito, 2019-03-27:
Same here.

**Server:**
C:\Users\jhasb>enip_server SCADA=INT

**Client:**
C:\Users\jhasb>enip_client SCADA

**Server Output:**
03-27 23:37:57.080   enip_srv enip.lgx ERROR    process    EtherNet/IP CIP error at 42 total bytes:
b'\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\xb2\x00\x1a\x00R\x02 \x06$\x01\x05\x9d\x0c\x00L\x04\x91\x05SCADA\x00\x01\x00\x01\x00\x01\x00'
-----------------------------------------------------------------------------------------------------------------------------------------------^ (byte 42)

03-27 23:37:57.080   enip_srv enip.srv ERROR    enip_srv_t Failed request: {
    "request.enip.command": 111,
    "request.enip.length": 42,
    "request.enip.session_handle": 2386738321,
    "request.enip.status": 0,
    "request.enip.sender_context.input": "array('B', [48, 0, 0, 0, 0, 0, 0, 0])",
    "request.enip.options": 0,
    "request.enip.input": "array('B', [0, 0, 0, 0, 0, 0, 2, 0, 0, 0, 0, 0, 178, 0, 26, 0, 82, 2, 32, 6, 36, 1, 5, 157, 12, 0, 76, 4, 145, 5, 83, 67, 65, 68, 65, 0, 1, 0, 1, 0, 1, 0])",
    "request.enip.CIP.send_data.interface": 0,
    "request.enip.CIP.send_data.timeout": 0,
    "request.enip.CIP.send_data.CPF.count": 2,
    "request.enip.CIP.send_data.CPF.item[0].type_id": 0,
    "request.enip.CIP.send_data.CPF.item[0].length": 0,
    "request.enip.CIP.send_data.CPF.item[1].type_id": 178,
    "request.enip.CIP.send_data.CPF.item[1].length": 26,
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.service": 82,
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.path.size": 2,
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.path.segment[0].class": 6,
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.path.segment[1].instance": 1,
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.priority": 5,
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.timeout_ticks": 157,
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.length": 12,
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.request.input": "array('B', [76, 4, 145, 5, 83, 67, 65, 68, 65, 0, 1, 0])",
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.route_path.size": 1,
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.route_path.segment[0].port": 1,
    "request.enip.CIP.send_data.CPF.item[1].unconnected_send.route_path.segment[0].link": 0,
    "request.addr": [
        "127.0.0.1",
        51612
    ]
}
03-27 23:37:57.098   enip_srv enip.srv ERROR    enip_srv_t EtherNet/IP error at 94 total bytes:
b'o\x00*\x00\x91\xbcB\x8e\x00\x00\x00\x000\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\xb2\x00\x1a\x00R\x02 \x06$\x01\x05\x9d\x0c\x00L\x04\x91\x05SCADA\x00\x01\x00\x01\x00\x01\x00'
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------^ (byte 66)

Failed with exception:
Traceback (most recent call last):
  File "c:\program files\python36\lib\site-packages\cpppo\server\enip\main.py", line 817, in enip_srv_tcp
    if enip_process( addr, data=data, **kwds ):
  File "c:\program files\python36\lib\site-packages\cpppo\server\enip\logix.py", line 777, in process
    data.response               = dotdict( data.request )
  File "c:\program files\python36\lib\site-packages\cpppo\dotdict.py", line 79, in __init__
    self.update( *args, **kwds )
  File "c:\program files\python36\lib\site-packages\cpppo\dotdict.py", line 85, in update
    self.__setitem__( key, val )
  File "c:\program files\python36\lib\site-packages\cpppo\dotdict.py", line 158, in __setitem__
    target[rest]        = value
  File "c:\program files\python36\lib\site-packages\cpppo\dotdict.py", line 158, in __setitem__
    target[rest]        = value
  File "c:\program files\python36\lib\site-packages\cpppo\dotdict.py", line 158, in __setitem__
    target[rest]        = value
  [Previous line repeated 1 more time]
  File "c:\program files\python36\lib\site-packages\cpppo\dotdict.py", line 153, in __setitem__
    target          = eval( mine, {'__builtins__':{}}, self )
  File "<string>", line 1, in <module>
NameError: name 'item' is not defined

[... 23 lines cut ...]
  File "c:\program files\python36\lib\site-packages\cpppo\dotdict.py", line 153, in __setitem__
    target          = eval( mine, {'__builtins__':{}}, self )
  File "<string>", line 1, in <module>
NameError: name 'item' is not defined

**Client** Output:
Traceback (most recent call last):
  File "c:\program files\python36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\program files\python36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python36\Scripts\enip_client.exe\__main__.py", line 9, in <module>
  File "c:\program files\python36\lib\site-packages\cpppo\server\enip\client.py", line 1630, in main
    fragment=fragment, printing=printing, timeout=timeout )
  File "c:\program files\python36\lib\site-packages\cpppo\server\enip\client.py", line 1406, in process
    transactions                = list( self.results( operations=operations, **kwds ))
  File "c:\program files\python36\lib\site-packages\cpppo\server\enip\client.py", line 1398, in results
    for idx,dsc,req,rpy,sts,val in self.operate( operations, **kwds ):
  File "c:\program files\python36\lib\site-packages\cpppo\server\enip\client.py", line 1393, in operate
    for idx,dsc,req,rpy,sts,val in harvested:
  File "c:\program files\python36\lib\site-packages\cpppo\server\enip\client.py", line 1283, in validate
    for index,descr,request,reply,status,val in harvested:
  File "c:\program files\python36\lib\site-packages\cpppo\server\enip\client.py", line 1270, in pipeline
    complete, requests )
AssertionError: Communication ceased before harvesting all pipeline responses:   0/  1

mjbrisebois, 2019-07-07:
This is now fixed in version 4.0.4

closed 2019-07-07 by mjbrisebois