pjkundert/cpppo #72: EtherNet/IP Server not handling requests from clients

bobbuilder123, 2019-06-20:
For the PoC, I am using Python 3.5.0
Command for the server: "python.exe -m cpppo.server.enip -v SINTARRAY=SINT[1000]"

The cpppo client *works* as I am able to fetch the first 20 elements from the array.
Command for the client: "python.exe -m cpppo.server.enip.client -v --print SINTARRAY[0-20]"
Answer: "SINTARRAY[0][  0-20 ]+  0 == [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]: 'OK'"

Both cpppo and Pylogix clients are able to fetch info from the real PLC, but Pylogix is not able to fetch info from the cpppo server. NOTE: "We also tried writing with a PLC (ControlLogix) on the cpppo server with no success."

PoC:
```
from pylogix import PLC
with PLC() as comm:
	comm.IPAddress = 'REDACTED_LOOPBACKIP'
	value = comm.Read('SINTARRAY[0]',20)
	print(value)
```
Error server-side:

> 06-20 16:29:21.065   enip_srv enip.srv NORMAL   enip_srv   EtherNet/IP Server enip_30686 begins serving peer ('REDACTED_IP', 30686)
> 06-20 16:29:21.081   enip_srv enip.dev ERROR    request    EtherNet/IP CIP error at 6 total bytes:
> b'T\x02 \x06$\x01\n\x0e\x02\x00\x00 \x01\x00\x00 \xc8\xe17\x13*\x00\x00\x00\x03\x00\x00\x004\x12 \x00\xffC\x01@ \x00\xffC\xa3\x03\x01\x06 \x02$\x01'
> -----------------^ (byte 6)
> 
> 06-20 16:29:21.092   enip_srv enip.dev NORMAL   request    (0x9999,  1) UCMM Command 0x006f SendRRData failed with Exception: Unconnected Send message with absent or empty request
> Request: {
>     "addr": [
>         "REDACTED_IP",
>         30686
>     ],
>     "enip.status": 0,
>     "enip.session_handle": 249184597,
>     "enip.CIP.send_data.interface": 0,
>     "enip.CIP.send_data.timeout": 0,
>     "enip.CIP.send_data.CPF.count": 2,
>     "enip.CIP.send_data.CPF.item[0].type_id": 0,
>     "enip.CIP.send_data.CPF.item[0].length": 0,
>     "enip.CIP.send_data.CPF.item[1].type_id": 178,
>     "enip.CIP.send_data.CPF.item[1].length": 48,
>     "enip.CIP.send_data.CPF.item[1].unconnected_send.request.get_attributes_all": true,
>     "enip.CIP.send_data.CPF.item[1].unconnected_send.request.input": "array('B', [84, 2, 32, 6, 36, 1, 10, 14, 2, 0, 0, 32, 1, 0, 0, 32, 200, 225, 55, 19, 42, 0, 0, 0, 3, 0, 0, 0, 52, 18, 32, 0, 255, 67, 1, 64, 32, 0, 255, 67, 163, 3, 1, 6, 32, 2, 36, 1])",
>     "enip.CIP.send_data.CPF.item[1].unconnected_send.request.service": 84,
>     "enip.CIP.send_data.CPF.item[1].unconnected_send.request.path.segment[0].class": 6,
>     "enip.CIP.send_data.CPF.item[1].unconnected_send.request.path.segment[1].instance": 1,
>     "enip.CIP.send_data.CPF.item[1].unconnected_send.request.path.size": 2,
>     "enip.sender_context.input": "array('B', [0, 0, 0, 0, 0, 0, 0, 0])",
>     "enip.command": 111,
>     "enip.length": 64,
>     "enip.options": 0
> }
> Traceback (most recent call last):
>   File "C:\REDACTED_PATH\cpppo\server\enip\device.py", line 1282, in request
>     CM.request( unc_send )
>   File "C:\REDACTED_PATH\cpppo\server\enip\device.py", line 1917, in request
>     MR.request( data.request )
>   File "C:\REDACTED_PATH\cpppo\server\enip\logix.py", line 237, in request
>     return target.request( data )
>   File "C:\REDACTED_PATH\cpppo\server\enip\device.py", line 1899, in request
>     "Unconnected Send message with absent or empty request"
> AssertionError: Unconnected Send message with absent or empty request
> 
> 06-20 16:29:21.097   enip_srv enip.srv WARNING  enip_srv_t Expected EtherNet/IP response encapsulated message; none found
> 06-20 16:29:21.098   enip_srv enip.srv WARNING  enip_srv_t Session ended (server EtherNet/IP status: 0x08 == 8)
> 06-20 16:29:21.099   enip_srv enip.srv NORMAL   enip_srv_t enip_30686 done; processed   2 requests over   116 bytes/  116 received (0 connections remain)

Error client-side:

> Traceback (most recent call last):
>   File "C:\REDACTED_PATH\test.py", line 16, in <module>
>     value = comm.Read('SINTARRAY[0]',30)
>   File "C:\REDACTED_PATH\pylogix\eip.py", line 93, in Read
>     return _readTag(self, tag, count, datatype)
>   File "C:\REDACTED_PATH\pylogix\eip.py", line 203, in _readTag
>     if not _connect(self): return None
>   File "C:\REDACTED_PATH\pylogix\eip.py", line 797, in _connect
>     sts = unpack_from('<b', retData, 42)[0]
> struct.error: unpack_from requires a buffer of at least 1 bytes

Thank you for the help and your time.

closed 2019-06-25 by bobbuilder123

karager09, 2020-05-08:
Hi,

I constantly receive the same error.
Have you got any tips how to resolve that?

bobbuilder123, 2020-05-08:
@karager09  Yes the cipppo server uses a connection "connected" while PLC uses by default "connectionless". Check that option on the PLC. Does it answer your question? I have completed my project a while ago already :p

karager09, 2020-05-09:
Well, my case is slightly different.
I have some module that is working with real PLC (eg. CompactLogix), but when I try to connect it to cpppo server, it gives me error "Unconnected Send message with absent or empty request" on server side (the same as you) and response to client is 0x08 0x00 0x00 0x00.
(I have tried TCP and UDP on client side but with no success).
I can change some messages send from module if that would help (have access to code), but I'm new to this and don't know what.

Thanks for your help 😄