pjkundert/cpppo #73: Unable to read/write single attribute.

tobiasht, 2019-06-24:
I am trying to connect to a simple eip device that supports "Get and Set Attribute Single". I am able to connect to the device and read some information from it, using the command below:
`py -3.5 -m cpppo.server.enip.client --list-identity --list-services -a 192.168.100.203 -t 10`

The problem is that i am not able to read any attributes from the device. My ultimate goal is to be able to read and write 12 elements from class 0x95, instance 1, attribute 1, but first i am just trying to get some data from the device that i know exist. 
(I am using EnIPExplorer to verify that the data i am trying to access actually exists and are accessible)
```
>py -3.5 -m cpppo.server.enip.get_attribute -a 192.168.100.203 '@0x95/1/1
Mon Jun 24 14:36:19 2019:   0: Single G_A_S      '@0x95/1/1 == None
```
```
>py -3.5 -m cpppo.server.enip.get_attribute -a 192.168.100.203 '@800/1/1
Mon Jun 24 14:37:20 2019:   0: Single G_A_S      '@800/1/1 == None
```

I have also tried to read other classes, with no luck. My response is always "None". Any help would be much appreciated.

pjkundert, 2019-06-24:
I would recommend adding some -v options, to see what there error is.

tobiasht, 2019-06-24:
```
>py -3.5 -m cpppo.server.enip.get_attribute -a 192.168.100.203 '@800/1/1 -v
06-24 15:57:06.347 MainThread enip.cli NORMAL   __init__   Connect:  Success in   0.053s/  5.000s
Mon Jun 24 15:57:06 2019:   0: Single G_A_S      '@800/1/1 == None
06-24 15:57:06.371 MainThread enip.get NORMAL   main         1 requests in   0.023s at pipeline depth  0;  42.921 TPS
```

closed 2019-06-29 by tobiasht