pjkundert/cpppo #76: How to read tag continiously?

GitHubAndDance, 2019-08-13:
Hi ,

How would I most effectively check tag continuously and react to certain value: 

like code below but the correct version :) 
```
tags = [' x[10]']
while 1:
    with client.connector as conn:
        for index,descr,op,reply,status,value in conn.pipeline(
                operation=client.parse_operations ( tags ), depth=2`:
            test = ("%s" % ( value ))
    if test == '[1]':
        print(something)

```

pjkundert, 2019-08-13:
I'd look at all the `poll_example*.py` examples in cpppo/server/enip for a start. They basically do what you want, I think.

TrailBlazerAI, 2019-10-29:
I have a similar requirement (continuous polling of tags). So do the routines in poll_example*.py allow pipelining and multiple requests? I am seeing a big difference in performance between some of the examples and thruput.py for example.