pjkundert/cpppo #83: Error reading strings from Omron PLC

TimGuiteDiamond, 2020-03-04:
Hi, thanks for creating this really cool module! It seems to work really well but I am having some issues reading from strings, some seem to work but not others?

I am using the `proxy_simple` interface for ease of use.

Some strings work:
```python
from cpppo.server.enip.get_attribute import proxy_simple
>>> host = "###.###.###.###"
>>> x, = proxy_simple(host).read(("V1.Desc"), printing=True)
         V1.Desc == Valve-01
```

Others give me the following error stream:
```python>>> x, = proxy_simple(host).read(("PLC_Name"), printing=True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/server/enip/get_attribute.py", line 437, in read
    for val,(sts,(att,typ,uni)) in reader:
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/server/enip/get_attribute.py", line 614, in read_details
    depth=self.depth, multiple=self.multiple, timeout=self.timeout )):
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/server/enip/client.py", line 1431, in operate
    for idx,dsc,req,rpy,sts,val in harvested:
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/server/enip/client.py", line 1296, in pipeline
    col		= next( harvester )
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/server/enip/client.py", line 1228, in harvest
    issued, self.collect( timeout=timeout )): # must be "lazy" zip!
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/server/enip/client.py", line 1154, in collect
    response,elapsed= await_response( self, timeout=timeout )
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/server/enip/client.py", line 911, in await_response
    for response in cli: # if StopIteration raised immediately, defaults to {} signalling completion
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/server/enip/client.py", line 562, in __next__
    for mch,sta in engine:
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 647, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 1228, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 647, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 1228, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 647, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 1228, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 647, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 1228, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 647, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 1228, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "/scratch/user/pipenv/pycomm-OV4KsteT/lib/python3.7/site-packages/cpppo/automata.py", line 612, in run
    "%s detected no progress before finding acceptable symbol" % ( self )
AssertionError: ((drop)) detected no progress before finding acceptable symbol
```

It seems to be fairly arbitrary which strings work and which ones don't. As far as I am aware they are all set up the same way on the PLC and I can indeed read them all using a different connection through our control system.

Any advice?

Thanks!

GraninDB, 2020-10-04:
Hello!
Try to comment code in automata.py
```
            #assert crumb not in seen, \
            #    "%s detected no progress before finding acceptable symbol" % ( self )
```

Philippe117, 2022-03-02:
Hi there, I can attest that the fix suggested by @GraninDB is working like a charm with my NJ101-1000 controller.
Thank you

GraninDB, 2022-03-02:
Hi!
Yes, You can.

GraninDB, 2022-03-02:
Sorry, I misunderstood your letter. Thanks for your message. I will try to
mark it in GIT