pjkundert/cpppo #85: ERROR TYPE modi_code= types.CodeType( *modi_args )

crocodile62, 2020-04-14:
I get this erro from PyCharme debug and it is not importing the libray.
I have no clue about it.
I get the same error launching python3 -m cpppo.server.enip

I think something is wrong in my installation, somebody could help me?
thanks

My environment is Mac PowerBook with Catalina and python 3.8 with PyCharme 
Thanks a lot for your help
this is the terminal response to   python3 -m cpppo.server.enip
traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 184, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 110, in _get_module_details
    __import__(pkg_name)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/cpppo/__init__.py", line 29, in <module>
    from .automata import *
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/cpppo/automata.py", line 33, in <module>
    from . import misc
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/cpppo/misc.py", line 204, in <module>
    change_function( __normal, co_filename=logging._srcfile )
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/cpppo/misc.py", line 163, in change_function
    modi_code			= types.CodeType( *modi_args )
TypeError: an integer is required (got type bytes)

 modi_code			= types.CodeType( *modi_args )
TypeError: an integer is required (got type bytes)

kdorsel, 2020-04-21:
I got the same error on Python 3.8.

It works fine on Python 2.7, and I imagine it also works fine on 3.7.

```
Python 3.8.2 (default, Feb 26 2020, 02:56:10) 
[GCC 7.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from cpppo.server.enip.get_attribute import proxy as plc
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/gastronomous/gastro/env/lib/python3.8/site-packages/cpppo/__init__.py", line 29, in <module>
    from .automata import *
  File "/home/gastronomous/gastro/env/lib/python3.8/site-packages/cpppo/automata.py", line 33, in <module>
    from . import misc
  File "/home/gastronomous/gastro/env/lib/python3.8/site-packages/cpppo/misc.py", line 204, in <module>
    change_function( __normal, co_filename=logging._srcfile )
  File "/home/gastronomous/gastro/env/lib/python3.8/site-packages/cpppo/misc.py", line 163, in change_function
    modi_code			= types.CodeType( *modi_args )
TypeError: an integer is required (got type bytes)
```

EarlAlnot, 2020-04-22:
Got the same error and managed to solve it by adding changing line 143. Seems types.CodeType got a new attribute (at least in a 3.8.x release).

Was: 
`        attrs		       += [ "co_kwonlyargcount" ]`

Now:
```
        attrs		       += [ "co_posonlyargcount" ,         # <-- Added
            		          "co_kwonlyargcount" ] 
```

First time posting and not so proficient with Github, so please have merci.

kdorsel, 2020-04-26:
Awesome, that fixed it indeed. For those looking:

https://github.com/pjkundert/cpppo/blob/272327c6d3be9f523358e73683b4862f6756b5a3/misc.py#L143

pjkundert, 2020-05-15:
Python 3.7/3.8 compatibility is now complete in version 4.0.5; pushed to pypi

closed 2020-05-15 by pjkundert

barthelmeh, 2023-06-19:
I'm still getting an error on the same line: _TypeError: code() argument 13 must be str, not int_
Unsure how to fix this.

EarlAlnot, 2023-06-19:
@barthelmeh 
Which version of python do you use and which error did you get?
Can you post full error as above?