pjkundert/cpppo #86: Open a persistent connector to fanuc robot

kdorsel, 2020-04-28:
I have a Fanuc Robot R30iB+ which I'm trying to connect to using:

```
from cpppo.server.enip.get_attribute import proxy as fanuc

host = "192.168.0.10"
param = "@0x6c/0x01/0x01"
fanuc = plc(host=host)

try:
    with fanuc:
        result = fanuc.read(param, checking=True)
        print(list(result))
except Exception as e:
    print(param, e)
```

This give me a return status 4 which is path segment error. The fanuc robot has eth/ip error that says the adapter is idle and thus not connected. So to me it seems that the problem is that there is no connection being opened to the robot or the connection is not maintained.

If I do I successfully get the information of the robot, but the controller still shows adapter not connected.
```
fanuc.open_gateway()
fanuc.identity
```

`list_identity()` seems to work, but I cannot open a persistent connection. Is there a way to do this? How do I correctly open a connection to read values?

pjkundert, 2020-09-16:
I've recently pushed up a new version 4.1.0, which now supports persistent "Forward Open" connections.  There is now a `proxy_connected` class in `cpppo.server.enip.get_attribute`.  Perhaps try that?  It will open up a persistent connection to a specific Object, and issue subsequent get_attribute_single requests to that CIP Object...

robin-gdwl, 2020-10-13:
@kdorsel did you succeed in establishing a persistent connection? 
If so: how did you do it ? I am facing the same problem.

kdorsel, 2020-10-13:
> @kdorsel did you succeed in establishing a persistent connection?
> If so: how did you do it ? I am facing the same problem.

I found this other EIP code which I used. https://codeberg.org/paperwork/pyenip